embassy-rs/embassy #1326: embassy-rp should provide PWM drivers similar to rp2040-hal

dreamsmasher, 2023-04-04:
Hi everyone, I'm trying to use `embassy-rp` for a synthesizer project to drive some pins using PWM. It seems that the rp2040 ecosystem is a bit fractured, with embassy preferring `embassy-rp` over `rp2040-hal`. This is a bit frustrating as `rp2040_hal` provides a high-level [PWM driver](https://docs.rs/rp2040-hal/latest/rp2040_hal/pwm/index.html) with typesafe channel definitions and convenience functions, but doesn't work with `embassy-time` due to the lack of a time driver impl. On the other hand, `embassy-rp` only exposes the raw PWM control registers from `rp2040-pac2`, without abstractions for configuring channels or binding them to pins.

It's my understanding that `embedded-hal` 1.0 will not include its `PWM` trait, but this isn't required - a simple, higher-level interface to the RP2040's PWM hardware would greatly improve UX and bring `embassy-rp` closer to feature parity with the other HAL crate.

Dirbaio, 2023-04-06:
Yes, this is unimplemented, but definitely wanted. 

PRs welcome. Happy to mentor or answer questions if you want to try implementing it.

dreamsmasher, 2023-04-13:
I have a prototype driver working as part of my project. It's a bit dirty wrt. extensive macro ~ab~use, and missing some functionality like interrupts. 

My main focus is on my synth firmware currently but I'd be glad to upstream the work once things are in a releasable state. If anyone else is interested in taking this on in the meantime, I can give access to the repo. Thanks!

Dirbaio, 2023-04-23:
#1387

Dirbaio, 2023-04-24:
done in #1387

closed 2023-04-24 by Dirbaio

dreamsmasher, 2023-04-24:
yay!!! thank you!