embassy-rs/embassy #1341: Race in STM32 v2 I2C DMA TX

ericyanush, 2023-04-07:
It seems there is a subtle race in the DMA TX I2c routine.

Steps to recreate:
- Initiate a vectored write on a v2 I2c peripheral (tested with STM32L431RC) that has DMA configured.
  - First transfer is small (2 bytes in my example; the address to write to on an I2c FRAM device)
  - Second transfer is large (in the attached example it's 1200bytes, but this happens in any transfer that is over 256 bytes).
- See the second transfer results in a panic on the call to master_continue AFTER the last chunk has been queued, but before the I2c EV irq has fired. 

I've attached a gist with a patch to show the log statements I added to narrow down the issue, as well as a sample of the output (which is 100% re-creatable).

From what I've seen (and from my rough understanding of how the wakers/task storage works), it appears that on the last chunk's transfer being completed by the DMA peripheral, it fires it's IRQ to indicate that the transfer has completed. Since the DMA Transfer object was created/registered from the same task as the actual poll_fn in the DMA write implementation, the task is polled again, where the poll_fn in the I2c dma write is polled, however, this happens before the I2c peripheral has triggered it's EV Irq, which updates the number of chunks transferred, so in this instance the poll_fn triggers another master_continue call, which panics, since it is sent a length of 0.

https://gist.github.com/ericyanush/c362e883a0ded1a367e9d5d44c8f8852

xoviat, 2023-04-15:
In terms of a fix I see two options:
1. Track whether the EV irq has completed before reading `state.chunks_transferred`.
2. Move to `remaining_len` as the tracking variable. 

Looking at the code, I can't see a reason why `chunks_transferred` needs to be independently tracked when `remaining_len` is available. I don't have access to an l4 at this point, but if you want I can propose changes that you can try to see if they resolve the problem.

Dirbaio, 2023-04-19:
@ericyanush can you confirm whether #1370 fixes your issue? Thanks!

closed 2023-04-19 by bors[bot]

ericyanush, 2023-04-21:
I can confirm that #1370 fixes the issue I encountered! Thanks @xoviat for getting that fixed so quickly!

commit f5216624 (2023-04-19): stm32/i2c: fix races when using dma.

fixes #1341.