embassy-rs/embassy #1343: Use STM32 SPI  as Neopixel WS2812b driver

smeenka, 2023-04-09:
To use the spi device for neopixel control it would be nice if the SPI device would support:
* a new constructor (new_neopixels) which only uses the Mosi pin, all other pins are None
* Is able to set the bitsize (dff register) to 5 bits.

Then with a clock of 4 Mhz, each spi transfer will clock out one bit for the neopixel, with an exact clock of 800 Khz.
Combined with the DMA transfer, it is an ideal way of controlling a neopixel string.

I will create a fork to implement this requested behavior from the staging branch, including an example file how to use it, for the STM32G070 (nucleo-g070RB board)

Dirbaio, 2023-04-09:
> a new constructor (new_neopixels) which only uses the Mosi pin, all other pins are None

Sounds good! nrf already has support for that, make sure to keep the API and function names consistent https://github.com/embassy-rs/embassy/blob/master/embassy-nrf/src/spim.rs#L103-L110

> Is able to set the bitsize (dff register) to 5 bits.

Sounds good as well! A field in `config` would work.

smeenka, 2023-04-10:
Here is my pull request. https://github.com/embassy-rs/embassy/pull/1345

I was not able to set the bitsize in the config. As it is tighly coupled to the datasize. On each write the datasize is set, depending on the data type. I was not sure If I could change this behavior

Dirbaio, 2024-06-02:
done in https://github.com/embassy-rs/embassy/pull/1345.

but see https://github.com/embassy-rs/embassy/issues/2902

closed 2024-06-02 by Dirbaio