embassy-rs/embassy #1354: nrf/ppi: add lifetimes to Task, Event.

Dirbaio, 2023-04-11:
Currently, you can:
- Create some driver
- Obtain a Task/Event from it
- Drop the driver
- Keep using the Task/Event

This might conflict with further uses of the peripheral by another driver which doesn't expect the user to be directly triggering tasks.
 
Fix is:
- Add lifetimes `Task<'d>`, `Event<'d>`
- Drivers returning tasks/events tie that lifetime to their `'d`
- PPI channels enforce they don't outlive the tasks/events.

We should check if this runs into self-referential issues when trying to store, e.g a Timer and a Ppi using its tasks in the same struct. I think it should be fine.

akshataj96, 2023-06-10:
Hi, I am a student looking to contribute in open source projects. 
If the issue is still available, can i start looking into this? @Dirbaio

ilikepi63, 2023-06-29:
Hey! 

Not entirely sure if this answers the above Issue, but I gave it a shot: https://github.com/embassy-rs/embassy/pull/1600

I surmise that the task would take the lifetime of the reference of the `self` argument, such as in here: 

```
    pub fn task_start(&self) -> Task {
                                   ^ here 

        Task::from_reg(&T::regs().tasks_start)
    }
```

Dirbaio, 2023-06-29:
`pub fn task_start(&self) -> Task` is equiavalent to `pub fn task_start<'a>(&'a self) -> Task<'a>` due to the [lifetime elision](https://doc.rust-lang.org/nomicon/lifetime-elision.html) rules. This means that the `&self` borrow must stay active as long as the `Task` exists. This'll prevent calling more `&mut` methods in the driver. I think this'll be too restrictive, there's probably existing code that does it.

What we want is

```rust
impl<'d, ...> Whatever<'d, ...> {
   pub fn task_start(&self) -> Task<'d> {...}
}
```

This allows creating the `Task` and then calling more `&mut` methods on the driver. It forbids destroying the driver and keep using the `Task`s, which is what we wanted to avoid.

ilikepi63, 2023-06-30:
Hey @Dirbaio, just a quick question: 

In the creation methods regarding PpiGroups, do the Tasks and Events related to the PpiGroup also inherit their lifetimes from the PpiGroup? 

The reason that I am asking is that when I retrieve a "disable all" tasks from a group, does that tasks share the same lifetime as the overall group? 

If not, then what is the lifetime of the Task? Would it potentially be static in this specific scenario?

Dirbaio, 2023-06-30:
yes, same as other drivers. You create the PpiGroup driver, get the task out of it. The task has a lifetime that prevents outliving the PpiGroup.

Dirbaio, 2024-03-03:
done in https://github.com/embassy-rs/embassy/pull/1600

closed 2024-03-03 by Dirbaio