embassy-rs/embassy #1386: HardFault on some stm32 devices.

TTTiNNNy, 2023-04-20:
Hi, its looks like some stm32f103(blue pill) and stm32f(401)/(411) dont properly work under embassy, but works fine under stm cube code generation. I build blinky example from stm32f1 and stm32f4 and try run it on my boards, but on some of them I fall into HardFault handler before main start(I put breakpoint on main func). One of the the 2 blue pill boards work fine and all of 2 black pills boards fall into HardFault. Any idea?
Backtrace:
```
#0  0x0800df82 in cortex_m_rt::HardFault_ (ef=0x0) at src/lib.rs:1046
#1  <signal handler called>
#2  0x00000000 in ?? ()

```

Dirbaio, 2023-04-20:
These chips are definitely supported, they should work.

Most common cause for immediate hardfault is specifying the wrong chip name in the `embassy-stm32` dependency in `Cargo.toml`, or specifying the wrong RAM size in `memory.x` if you're supplying it manually.

TTTiNNNy, 2023-04-22:
It's seems like I really had wrong chip name in Cargo.toml's features of embassy-stm32 dependency. I also notice, that memory.x should be use directly, like for nrf52. Thanks for help!

closed 2023-04-22 by TTTiNNNy