embassy-rs/embassy #1389: Mixing crates.io and git versions of embassy crates causes wrong delays

JuliDi, 2023-04-22:
The `embassy_time::Delay` gives wrong delay times when using embassy-time from crates.io while using embassy-stm32 from git like this:
Cargo.toml
```toml
embassy-time = { version = "0.1", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-2_000_000", "unstable-traits", "nightly"] }
embassy-stm32 = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["nightly", "defmt", "unstable-pac", "stm32l432kc", "time-driver-any", "exti", "unstable-traits"]  }
```

When I set the embassy-time feature `tick-hz-2_000_000` (to set the TICK_HZ correctly here https://github.com/embassy-rs/embassy/blob/master/embassy-time/src/tick.rs#L110) it will not be 2 MHz in embassy-stm32 here https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/time_driver.rs#L156 but stay at 1 MHz.

The documentation says that this feature should not be set by the end-user but rather by the timer driver. However, the embassy-stm32 driver reads the TICK_HZ defined by this exact feature. So from what I see this is still the way to change it.

`cargo tree` tells me that there are two different versions of embassy-time used, since embassy-stm32 ist not available on crates.io (hence the git version is used). But the version where I set the tick in my Cargo.toml was in fact the one version from crates.io.

So one way to fix it is using embassy-time from the git repo instead of crates.io.
So far, so consistent.

However, I cannot use embassy-executor from git because for some reason it won't compile then:
```
   |
23 | #[embassy_executor::main]
   |                     ^^^^ could not find `main` in `embassy_executor`
``` 

For now I am using it like this
```toml
embassy-executor = { version = "0.1", features = ["nightly", "defmt", "integrated-timers"] }
embassy-time = { version = "0.1", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-2_000_000", "unstable-traits", "nightly"] }
embassy-stm32 = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["nightly", "defmt", "unstable-pac", "stm32l432kc", "time-driver-any", "exti", "unstable-traits"]  }
```

Is there any way to make this more consistent and avoid wrong delays?
I guess it would suffice if I could tell cargo to enable the features for all versions of the crate, crates.io and git. But this is not possible afaik.

Dirbaio, 2023-04-23:
> However, I cannot use embassy-executor from git because for some reason it won't compile then:

enable features `arch-cortex-m` and `executor-thread` in `embassy-executor`. See the examples in the repo.

> Is there any way to make this more consistent and avoid wrong delays?

Don't mix crates.io and git.

Due to the way the `embassy-time` driver is structured, I don't think there's a way to make mixing transparently work. What we could do is add a `links` key to `Cargo.toml`, so that Cargo will refuse to link two copies of the crate.

JuliDi, 2023-04-23:
That's for the hint!
Interesting that it works without the features when using crates.io

I think refusing to link would be a good thing. Finding the cause of the issue took me quite a while and I don't see any good reason why one would want to link different versions.

plaes, 2023-11-29:
@Dirbaio Wasn't this issue fixed with embassy-time-1.0.5 by https://github.com/embassy-rs/embassy/pull/2074 ?

closed 2023-11-29 by Dirbaio