embassy-rs/embassy #1394: embassy-sync: Add RwLock

PegasisForever, 2023-04-25:
Something like [async-std's RwLock](https://docs.rs/async-std/latest/async_std/sync/struct.RwLock.html) would be extremely helpful for implementing async programs.

PegasisForever, 2023-04-25:
I think I can give a shot on this, I already got a prototype working. Is there anyway I can write tests for asynchronous primitives running on embassy's executor? Or I just create a normal test which runs the rwlock using tokio?

Dirbaio, 2023-04-25:
There's already some tests using `futures-test`: https://github.com/embassy-rs/embassy/blob/master/embassy-sync/src/channel.rs#L472

Using another executor is fine, embassy-sync is executor-independent and using embassy-executor would be annoying for tests because tasks are statically allocated.

PegasisForever, 2023-04-25:
Didn't saw that, thanks!

elpiel, 2023-04-26:
Awesome idea for a feature. I've wondered why there hasn't been anything like that so far.

Ragarnoy, 2023-05-31:
How hard to implement would this be for a beginner?

PegasisForever, 2023-06-01:
@Ragarnoy This is my current implementation so far: https://github.com/McMaster-Rocketry-Team/rust-monorepo/blob/master/vlfs/src/utils/rwlock.rs It works, however it is not fair, meaning too many readers can starve writers.

PR #3932 (2025-04-13): Add RwLock to embassy-sync
Fixes #1394

The implementation is very similar to `Mutex`
commit 025d9f6e (2025-04-13): Add RwLock to embassy-sync

Fixes #1394

---

For more details, open the [Copilot Workspace session](https://copilot-workspace.githubnext.com/embassy-rs/embassy/issues/1394?shareId=XXXX-XXXX-XXXX-XXXX).
closed 2025-04-13 by Dirbaio