embassy-rs/embassy #1413: Multiple Executor example to run tasks at different priority levels on RP2040.

overheat, 2023-04-29:
I have two different things need to manage, time critical motor/sensor control and networking. One fellow suggest using an InterruptExecutor for higher prio tasks with embassy.

I only found a example for nrf52840, [here](https://github.com/embassy-rs/embassy/blob/master/examples/nrf52840/src/bin/multiprio.rs
)

Need example for rp2040

closed 2023-05-08 by bors[bot]

overheat, 2023-05-19:
@fakusb Thanks for your PR.

I have a question that what if the high priority task is a hardware interrupt, not SWI, let's say I2C1 interrupt?
Can I modify just `SWI_IRQ_1` to `I2C1_IRQ` to let i2c2 interrupt everything else?

Dirbaio, 2023-05-19:
You can use any interrupt that's unused. If you are really using I2C1, then the I2C driver is using `I2C1_IRQ`, you can't use it for an executor at the same time.

overheat, 2023-05-19:
> You can use any interrupt that's unused. If you are really using I2C1, then the I2C driver is using `I2C1_IRQ`, you can't use it for an executor at the same time.

Then how to set different priority for different periphal? Let's say I don't want WiFi (spi or pio) interrupt motor driver (pwm)?

Dirbaio, 2023-05-19:
you control the priorities of tasks, not peripherals. If you spawn a task in a high-priority executor, everything that task does (including the peripheralsit manages) are high priority.

commit 43fd4f66 (2023-05-08): Merge #1435

1435: Added example for multi priority executors rp2040 r=Dirbaio a=fakusb

I added an example for multiple priorities of tasks on rp2040 by adjusting [examples/nrf52840/src/bin/multiprio.rs](https://github.com/embassy-rs/embassy/blob/master/examples/nrf52840/src/bin/multiprio.rs) .

This needs https://github.com/embassy-rs/rp-pac/pull/2 , and this commit also adds the 6 new interrupt handlers for software interrupts to embassy-rs.

We might need to change the git path for rp-pac in [embassy-rp/Cargo.toml](https://github.com/embassy-rs/embassy/compare/master...fakusb:rp2040-multip [... cut]
commit d0703f83 (2023-05-08): Merge #1435

1435: Added example for multi priority executors rp2040 r=Dirbaio a=fakusb

I added an example for multiple priorities of tasks on rp2040 by adjusting [examples/nrf52840/src/bin/multiprio.rs](https://github.com/embassy-rs/embassy/blob/master/examples/nrf52840/src/bin/multiprio.rs) .

This needs https://github.com/embassy-rs/rp-pac/pull/2 , and this commit also adds the 6 new interrupt handlers for software interrupts to embassy-rs.

We might need to change the git path for rp-pac in [embassy-rp/Cargo.toml](https://github.com/embassy-rs/embassy/compare/master...fakusb:rp2040-multip [... cut]
PR #1435 (2023-05-08): Added example for multi priority executors rp2040
I added an example for multiple priorities of tasks on rp2040 by adjusting [examples/nrf52840/src/bin/multiprio.rs](https://github.com/embassy-rs/embassy/blob/master/examples/nrf52840/src/bin/multiprio.rs) .

This needs https://github.com/embassy-rs/rp-pac/pull/2 , and this commit also adds the 6 new interrupt handlers for software interrupts to embassy-rs.

We might need to change the git path for rp-pac in [embassy-rp/Cargo.toml](https://github.com/embassy-rs/embassy/compare/master...fakusb:rp2040-multiprio-executor?expand=1#diff-47463ea35874 [... cut]