embassy-rs/embassy #1421: probe-run vs probe-rs-cli for nrf52840 Spim<TWISPI1> examples

ceekdee, 2023-05-02:
For nrf52840 LoRa examples (using Spim<TWISPI1>), the probe-run method still results in correct SPI operation.  For example:

- probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabi/release/lora_p2p_send

However, the probe-rs-cli method results in SPIM read failures (the first 2 read operations return all 0xff values, the second read operation resulting in a panic since the read 0xff value is an invalid index value):

- probe-rs-cli run --chip nRF52840_xxAA target/thumbv7em-none-eabi/release/lora_p2p_send

All the nrf52840 lora examples will fail using probe-rs-cli.  The stm32wl, stm32l0, and rp LoRa examples successfully run with probe-rs-cli.

ceekdee, 2023-05-02:
The RAK4631 (nRF52840/sx1262) is flashed through a set up using an nRF52840-DK with a SEGGER J-Link debugger/programmer.  The wiring from the nRF52840-DK to the RAK4631 is taken from the following resource:

https://devzone.nordicsemi.com/f/nordic-q-a/64261/programming-debugging-external-custom-nrf52840-boards-using-nrf52840-dk.

Perhaps this kludge is a bit much for probe-rs-cli?

ceekdee, 2023-06-29:
@Dirbaio - perhaps this issue can be closed when PR https://github.com/embassy-rs/embassy/pull/1595 is merged?

ceekdee, 2023-07-27:
Now that I have a test environment available again, neither probe-rs nor probe-rs-cli successfully sets up the nrf52840 lora examples for SPI using TWISPI1.  probe-run still sets up TWISPI1 successfully.

plaes, 2024-02-08:
@Dirbaio I think we can close this as:
a) LoRa examples were moved to lora-rs organization
b) probe-run has been deprecated.

closed 2024-02-08 by Dirbaio