embassy-rs/embassy #1427: stm32: rx dma ringbuf is flaky

Dirbaio, 2023-05-02:
steps to reproduce:
- checkout https://github.com/embassy-rs/embassy/commit/a21a2901acc839b512f40beeee4f77eb5ee8913d
- `cargo run --release --features stm32f103c8 --bin usart_rx_ringbuffered`
- after a few seconds, it stops transmitting data. No overrun error, no corrupted data.

sample output: https://gist.github.com/Dirbaio/49fae6ef02747bcdbd3ad2df5e5f6671

rmja, 2023-05-04:
I cannot reproduce this on f4.

Here is my local diff relative to current master:

<details>
<summary>Patch</summary>

```
diff --git a/tests/stm32/build.rs b/tests/stm32/build.rs
index 7ae31177..d3f1f39a 100644
--- a/tests/stm32/build.rs
+++ b/tests/stm32/build.rs
@@ -9,7 +9,11 @@ fn main() -> Result<(), Box<dyn Error>> {
     println!("cargo:rustc-link-arg-bins=--nmagic");
 
     // too little RAM to run from RAM.
-    if cfg!(any(feature = "stm32f103c8", feature = "stm32c031c6")) {
+    if cfg!(any(
+        feature = "stm32f429zi",
+        feature = "stm32f103c8",
+        feature = "stm32c031c6"
+    )) {
         println!("cargo:rustc-link-arg-bins=-Tlink.x");
         println!("cargo:rerun-if-changed=link.x");
     } else {
diff --git a/tests/stm32/src/bin/usart_rx_ringbuffered.rs b/tests/stm32/src/bin/usart_rx_ringbuffered.rs
index 9d75dbe5..066c0937 100644
--- a/tests/stm32/src/bin/usart_rx_ringbuffered.rs
+++ b/tests/stm32/src/bin/usart_rx_ringbuffered.rs
@@ -10,7 +10,7 @@ use defmt::{assert_eq, panic};
 use embassy_executor::Spawner;
 use embassy_stm32::interrupt;
 use embassy_stm32::usart::{Config, DataBits, Parity, RingBufferedUartRx, StopBits, Uart, UartTx};
-use embassy_time::{Duration, Timer};
+use embassy_time::{Duration, Timer, Instant};
 use example_common::*;
 use rand_chacha::ChaCha8Rng;
 use rand_core::{RngCore, SeedableRng};
@@ -35,9 +35,9 @@ mod board {
 }
 #[cfg(feature = "stm32f429zi")]
 mod board {
-    pub type Uart = embassy_stm32::peripherals::USART6;
-    pub type TxDma = embassy_stm32::peripherals::DMA2_CH6;
-    pub type RxDma = embassy_stm32::peripherals::DMA2_CH1;
+    pub type Uart = embassy_stm32::peripherals::USART2;
+    pub type TxDma = embassy_stm32::peripherals::DMA1_CH6;
+    pub type RxDma = embassy_stm32::peripherals::DMA1_CH5;
 }
 #[cfg(feature = "stm32wb55rg")]
 mod board {
@@ -64,7 +64,7 @@ mod board {
     pub type RxDma = embassy_stm32::peripherals::DMA1_CH2;
 }
 
-const DMA_BUF_SIZE: usize = 256;
+const DMA_BUF_SIZE: usize = 128;
 
 #[embassy_executor::main]
 async fn main(spawner: Spawner) {
@@ -90,12 +90,12 @@ async fn main(spawner: Spawner) {
         (p.PC4, p.PC5, p.USART1, interrupt::take!(USART1), p.DMA1_CH1, p.DMA1_CH2);
     #[cfg(feature = "stm32f429zi")]
     let (tx, rx, usart, irq, tx_dma, rx_dma) = (
-        p.PG14,
-        p.PG9,
-        p.USART6,
-        interrupt::take!(USART6),
-        p.DMA2_CH6,
-        p.DMA2_CH1,
[... 107 lines cut ...]
         if idles {
-            let micros = (random::<usize>() % 1000) as u64;
+            let micros = (random::<usize>() % 500) as u64;
             println!("Sleeping {}us", micros);
             port.flush().unwrap();
             thread::sleep(Duration::from_micros(micros));

```
</details>

Here are the tests that I have done. Note that I use usart2 instead of usart6. I have adjusted the interrupt priorities accordingly. The priorities does not seem to make any difference.

1) Loopback:
Simply short the rx and tx pin.
Build `embassy-stm32-tests` with `cargo build --bin usart_rx_ringbuffered --release --features=stm32f429zi`.
The output from `defmt` gives an indication that the test runs.

2) With host PC.
Wire rx pin to a serial port on a pc.
Build `embassy-stm32-tests` with `cargo build --bin usart_rx_ringbuffered --release --features=stm32f429zi`.
Run the build firmware.
Run `test-utils` with `cargo run com2 --idles`.
The output from `defmt` gives an indication that the test runs.

Setting a lower `DMA_BUF_SIZE` to, say 64 bytes gives me immediate overrun errors.

dragonnn, 2024-01-15:
I think I am getting the same issue, if I am recreating the Uart instance on each loop iteration it works fine, but if use it constantly it stops working after a few kB received, doesn't metter if the rx buffer is 256 or 512, it stops basically always at the same point. Sometimes it does recover after a few secs more but not always.
What is strange is that this does seam to only happen on USART2, my program uses USART6 too and even more data goes through it and it works perfectly fine. My MCU is STM32F401RET6.

rmja, 2024-01-15:
@dragonnn Could you try and assign the same interrupt priority to USART2 as you have on USART6, and verify that it still does not work?

rmja, 2024-01-15:
How about the dma priority? USART2 uses DMA1 and USART6 DMA2.

It is actually strange, as if not configured, USART2 and DMA1 should have default higher priority than USART6 and DMA2 according to table 38 in the resource manual.

I guess it is not a priority issue then.

dragonnn, 2024-01-15:
Hmmm
```
0.003969 INFO  current usart priority: P0
└─ <mod path> @ └─ <invalid location: defmt frame-index: 309>:0   
0.004005 INFO  current usart2 dma1 ch6 priority: P0
└─ <mod path> @ └─ <invalid location: defmt frame-index: 310>:0   
0.004042 INFO  current usart2 dma1 ch5 priority: P0
└─ <mod path> @ └─ <invalid location: defmt frame-index: 311>:0   
0.004117 INFO  current usart6 priority: P0
└─ <mod path> @ └─ <invalid location: defmt frame-index: 314>:0   
0.004153 INFO  current usart6 dma2 ch6 priority: P0
└─ <mod path> @ └─ <invalid location: defmt frame-index: 315>:0   
0.004189 INFO  current usart6 dma2 ch1 priority: P0
```
Does embassy zero out all priority?