embassy-rs/embassy #1441: Immediate Error::Overrun  when receiving data with embassy_stm32::usart::BufferedUart or RingBufferedUartRx

janschiefer, 2023-05-10:
I'm trying to communicate with a LTE modem and run into a couple of problems with the current UART implementations of embassy_stm32.

- When using BufferedUartRx or the new RingBufferedUartRx I run into an immediate Error::Overrun when testing in loopback mode when I send and receive from the same board in parallel. Even with very big rx_buffer and very little transmitted data. This applies to fill_buf() and read(&mut [u8]) methods.

- When I try receiving from a serial to UART converter in low speed (because I type in the text to send on my PC) I can receive some text, before I also receive Error::Overrun. 

- The consume(&self, usize) from BufferedUartRx / RingBufferedUartRx method seems to have no effect in preventing overruns.

- UartRx in DMA mode with read_until_idle() kinda works together with embassy time::with timeout, but this solution is quite cumbersome and suboptimal.

Tested on an old STM32F103C8T6 "BluePill" board with original STM32.

abelREK, 2023-09-12:
I have also replicated this issue on a STM32f405RG.

mxseev, 2023-12-20:
Perhaps the problem is in the default RCC configuration. I had the same thing on my STM32F411CEU6 (Black pill), but after manual RCC setup (at least enabling HSE) everything worked fine. My config:

```rust
    let mut config = Config::default();
    config.rcc.hse = Some(Hse {
        freq: Hertz(25_000_000),
        mode: HseMode::Oscillator,
    });
    config.rcc.pll_src = PllSource::HSE;
    config.rcc.pll = Some(Pll {
        prediv: PllPreDiv::DIV25,
        mul: PllMul::MUL200,
        divr: Some(PllRDiv::DIV2),
        divp: Some(PllPDiv::DIV2),
        divq: None,
    });
    config.rcc.ahb_pre = AHBPrescaler::DIV1;
    config.rcc.apb1_pre = APBPrescaler::DIV2;
    config.rcc.apb2_pre = APBPrescaler::DIV2;
    config.rcc.sys = Sysclk::PLL1_P;

    let p = embassy_stm32::init(config);
```

FourLeafTec, 2024-06-06:
Same issue on STM32F407VET6.