embassy-rs/embassy #1443: USB Host Pico

MarcoKonrad, 2023-05-10:
Hello everybody,

does anyone know if it is possible to make the RP Pico a usb host using embassy? If not, does anyone know a diffrent rust crate that does this?

I know there is a C Libary for that, but i want to avoid joining two Languages since i dont know how C and Embassy work together in embbeded coding.

MarcoKonrad, 2023-05-10:
Since @Dirbaio added this to the kind-features I assume it is currently not possible.

If anyone would have an idea for my problem, I would be highly thankful.

I do have a HID USB Interface that I want to send data to the pico which later should be send to a web api using the WiFi module on the pico.

I am new to  embedded coding and I am struggling a little bit with the whole stuff🙈

Dirbaio, 2023-05-10:
Currently `embassy-usb` is device-mode-only, yes. I'd love to support host mode someday too, but as far as I know no one is planning to work on it for now. PR's welcome :) 

There's a Rust USB Host implementation for STM32 [here](https://github.com/nviennot/turbo-resin/tree/main/src/drivers/usb) (Apache+MIT licensed) that looks neat, that we could use as a starting point. A lot of the logic is chip-independent.

MarcoKonrad, 2023-05-11:
@Dirbaio I looked a little bit into it, but have a hard time to understand the whole process. Do you have any ressources to help me understand how to develop a driver in general? How did you managed to get started in coding cyw43 for example?

vic1707, 2024-10-23:
Hi, do we have any updates on this?
I wanted to engage on a journey to learn embedded, found embassy and was very hyped to try it out, but I want to do a toy project that requires host support 😓.

Joeboy, 2025-10-25:
I see this issue is a year old now but FWIW as of now:

There's a draft PR [here](https://github.com/embassy-rs/embassy/pull/4321), not sure if anybody's still actively working on it. But, I did manage to check out the branch and get it working on a Pico 2 with a USB keyboard quite easily. [This](https://github.com/embassy-rs/embassy/issues/3295) looks to be the relevant ticket.

There's also, incidentally, usb-cotton-host, which says in its docs it's designed for platforms "like" embassy but seems to mostly target rtic. Wasn't obvious to me how to integrate it with embassy so I went back to the above WIP branch.

creeper6530, 2026-10-03:
> I think this issue can be closed now that [#5633](https://github.com/embassy-rs/embassy/pull/5633) has been merged.

Am I being dense, or has this not been released on crates.io yet? I am trying to implement the example usb_host_keyboard with `embassy-rp = {version = "0.10.0", features = […]}` and the compiler can't find the entire `host` module.