embassy-rs/embassy #1445: Add float example for RP2040?

overheat, 2023-05-11:
In RP2040 [datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) `2.8.3.2. Fast Floating Point Library`, there are optimized single-precision and double-precision floating point implementation.

Any example for it?

CBJamo, 2023-05-11:
Good news! As of #1382 the fast float in the rom is used by default.

overheat, 2023-05-17:
Yes, it works on core0.
But it's stop working when I put it into core1. I have no idea.

overheat, 2023-05-19:
@CBJamo Thanks.

An issue, I test `tests/rp/src/bin/float.rs` but it failed on my pico board.
Cloud you kindly test it?

overheat, 2023-05-20:
After enable "intrinsics" and "rom-v2-intrinsics" feature, room accesses is `141190`. 
But it still slower.

I change `let rad_f = (i as f32) * PI_F / 180.0;` to
```
        let rad_f = rom_data::float_funcs::fmul((i as f32), PI_F);
        let rad_f = rom_data::float_funcs::fdiv((rad_f), 180.0);
```
, then the time consuing reduced from 19ms to around 6ms.

overheat, 2023-05-21:
@pennae 
Yes, make sence!

After enable "intrinsics" and "rom-v2-intrinsics" feature, the performance lose 25%, compare to not enable them.
From my understanding enable means `using the ROM`, not enable means `doing soft-float maths`. 
What I am confused it that `using the ROM` slower than `doing soft-float maths`?

overheat, 2023-05-22:
@pennae 
Thanks! One more question related to float operation.

When I put heavy math compute task on rp2040 core1, it stuck the whole chip. Same heavy math compute task works on core0 is fine. Any hits for this issue?

overheat, 2023-05-22:
Yes, maybe. I will dig deeper and rise another issue.
Thanks.

closed 2023-05-22 by overheat