embassy-rs/embassy #1449: xtensa executor implementation can deadlock on multi-core systems

bugadani, 2023-05-13:
On multi-core systems, the [critical section implementation](https://github.com/esp-rs/esp-hal/blob/fa3627c1fda334ac89181ae293f762eab82bb307/esp-hal-common/src/lib.rs#L192) looks like the following: 
 - mask interrupts
 - acquire a cross-core lock
 - release cross-core lock
 - unmask interrupts

The [executor implementation](https://github.com/embassy-rs/embassy/blob/82f7e104d90a6628d1873017ea5ef6a7afb3b3f7/embassy-executor/src/arch/xtensa.rs#L78) relies on the `waiti` instruction to unmask interrupts, releasing the critical section without releasing the cross-core lock. This can lead to deadlocks where, after waiti ends the critical section, an interrupt handler tries to acquire the multi-core lock that hasn't been released. The following screenshot is an example of this deadlock:

[image]

Dirbaio, 2023-05-13:
I'm not familiar at all with xtensa I'm afraid.

- Does `waiti` really unmask interrupts? in ARM and RISCV it doesn't: it waits until an interrupt becomes pending, and then returns control. The interrupt doesn't get handled right then because interrupts are masked, it gets handled later when exiting the critical section.
- The `SIGNAL_WORK_THREAD_MODE` being a static is probably wrong in multicore, there should be one per executor...

bugadani, 2023-05-13:
> Does waiti really unmask interrupts?

I was going by what I read in the comments, but looking at the ISA, I'd say the answer is yes:

 > The WAITI instruction sets the current interrupt level in the PS.INTLEVEL register. In some
implementations it also powers down the processor’s logic, and waits for an interrupt. After
executing the interrupt handler, execution continues with the instruction following the WAITI.

 > I'm not familiar at all with xtensa I'm afraid.

cc @MabezDev then, maybe?

Dirbaio, 2023-05-13:
> The WAITI instruction sets the current interrupt level in the PS.INTLEVEL register.

wat

anyway, The ARM thread mode executor works by relying on `WFE/SEV` to signal events across cores, which seems wired like that in all multicore chips I've seen (?). This makes it possible to write a multicore-safe executor that runs on all ARM chips. 

If multicore xtensa requires to do more chip-specific stuff perhaps `esp-hal` should provide the executor impl, based on `embassy_executor::raw`, and then we rename `arch-xtensa` to `arch-xtensa-single-core`. 

Also I think riscv has the same problems, lol. (except `WFI` doesn't reenable interrupts)

Dirbaio, 2023-05-14:
With #1451 the deadlock is fixed, but the executor still doesn't properly support multi-core (ie you can't create two executors one in each core). Let's repurpose this issue to track that.

Also linking [relevant discussion on Matrix](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$AlG_eddW7ApR_HTfLJVcFa3UrcXcwYa2iLX8VB9TnhM?via=matrix.org&via=tchncs.de&via=mozilla.org)

bugadani, 2023-08-22:
Now that https://github.com/esp-rs/esp-hal/pull/723 exists for ESP32 and there is no hope on the ISA level to add multi-core support, maybe we can close this issue.

closed 2023-08-22 by bugadani

kokroo, 2025-10-26:
@bugadani I am a little confused. I saw your PR on esp-hal.
Does it mean that the embassy executor can spawn tasks on multiple cores now?