embassy-rs/embassy #1466: Difficulty Compiling for STM32

alexshirley, 2023-05-19:
So a bit of a n00b, but I'm running into troubles creating a new project

```bash
libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol WWDG
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol RTC
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol FLASH
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol RCC
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol EXTI0_1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol EXTI2_3
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol EXTI4_15
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol DMA1_CHANNEL1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol DMA1_CHANNEL2_3
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol DMA1_CHANNEL4_5
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol ADC1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM1_BRK_UP_TRG_COM
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM1_CC
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM3
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM6
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM7
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM14
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM15
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM16
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol TIM17
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol I2C1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol I2C2
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol SPI1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol SPI2
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol USART1
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol USART2
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol USART3_4
          libstm32_metapac-bd62a65356f59473.rlib(stm32_metapac-bd62a65356f59473.stm32_metapac.160da54d-cgu.4.rcgu.o) : error LNK2001: unresolved external symbol USB
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol Reset
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol NonMaskableInt
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol HardFaultTrampoline
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol MemoryManagement
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol BusFault
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol UsageFault
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol SVCall
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol DebugMonitor
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol PendSV
          libcortex_m_rt-e3b58d0a626900a1.rlib(cortex_m_rt-e3b58d0a626900a1.cortex_m_rt.2fb0f525-cgu.3.rcgu.o) : error LNK2001: unresolved external symbol SysTick
```

Here are my dependencies

```rust
[dependencies]
stm32-metapac = "7.0.0"
cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.1"
cortex-m-semihosting = "0.3.3"
panic-halt = "0.2.0"
embassy-executor = {version = "0.2.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "integrated-timers"] }
embassy-time = {version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-stm32 = {version = "0.1.0", git = "https://github.com/embassy-rs/embassy.git", features = ["defmt","stm32f070cb"]}
```

My kneejerk reaction is that `embassy-stm32` and `cortex-m-rt` are missing additional dependencies, or some additional toml settings, which I've tried to remediate by pulling them into my dependencies as well. However, I have no idea what seems to be wrong with my current setup.

alexshirley, 2023-05-19:
Quick update:

I decided to switch over to nightly, redownloaded the STM32 v6m compiler, and still got a very similar message, this time with the full context:

```bash
error: linking with `link.exe` failed: exit code: 1120
  |
  = note: "D:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX64\\x64\\link.exe" "/NOLOGO" "C:\\Users\\Me\\AppData\\Local\\Temp\\rustcN4WX28\\symbols.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.0.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.1.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.10.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.11.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.12.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.13.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.14.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.15.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.2.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.3.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.4.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.5.rcgu.o" 
"C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.6.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.7.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.8.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.build_script_build.81c1c68622bd59a9-cgu.9.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\embassy-stm32-58942b6ee258a7a6\\build_script_build-58942b6ee258a7a6.1859h6dhhcwkefk9.rcgu.o" "/LIBPATH:C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps" "/LIBPATH:C:\\Users\\Me\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\stm32-metapac-7.0.0/src/chips/stm32f070cb" "/LIBPATH:C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\build\\cortex-m-rt-2ebccca765b4c1e1\\out" "/LIBPATH:C:\\Users\\Me\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libstm32_metapac-9912149c59a1d024.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libcortex_m-d2eb502326c64cda.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libembedded_hal-1f47a0238fb4ac28.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libvoid-6b3aa605112187c1.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libnb-38e8662f8b5aab55.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libnb-d90d895c5489634b.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libvolatile_register-52a8f9c9d1491acf.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libvcell-061b6a584b814dc5.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libbare_metal-eafd81ca5e1206e6.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libcortex_m_rt-4faad5b4cc4830e8.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libquote-1c5a5b99f2b01435.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libproc_macro2-224ef92cf7a8e297.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps\\libunicode_ident-fb44f0087c2a31dc.rlib" "C:\\Users\\Me\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libproc_macro-9e71230232747fa7.rlib" "C:\\Users\\Me\\.rustup\\toolchains\\nightly-x86_64-pc-windows-m [... cut ...]
          libcortex_m_rt-4faad5b4cc4830e8.rlib(cortex_m_rt-4faad5b4cc4830e8.cortex_m_rt.85cf949c9bb5a779-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol PendSV
          libcortex_m_rt-4faad5b4cc4830e8.rlib(cortex_m_rt-4faad5b4cc4830e8.cortex_m_rt.85cf949c9bb5a779-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol SysTick
          C:\Users\Me\Documents\embed-rust\app\target\debug\build\embassy-stm32-58942b6ee258a7a6\build_script_build-58942b6ee258a7a6.exe : fatal error LNK1120: 38 unresolved externals

error: could not compile `embassy-stm32` (build script) due to previous error
```

alexshirley, 2023-05-19:
I updated

```rust
[package]
...
edition = "2021"
...
```

and that seemed to change _something_

```bash
error[E0433]: failed to resolve: unresolved import
 --> C:\Users\Me\Documents\embed-rust\app\target\thumbv6m-none-eabi\debug\build\embassy-stm32-fdf8868461f25813\out/_generated.rs:1:28806
  |
1 | ...: usart :: CtsPin , USART4 , PB7 , 4u8) ; pin_trait_impl ! (crate :: usb :: DmPin , USB , PA11 , 0u8) ; pin_trait_impl ! (crate :: usb :: DpPin , USB , PA12 , 0u8) ; dma_trait_impl ! (crate :... 
  |                                                                         ^^^
  |                                                                         |
  |                                                                         unresolved import
  |                                                                         help: a similar path exists: `stm32_metapac::usb`

error[E0433]: failed to resolve: unresolved import
 --> C:\Users\Me\Documents\embed-rust\app\target\thumbv6m-none-eabi\debug\build\embassy-stm32-fdf8868461f25813\out/_generated.rs:1:28868
  |
1 | ...ate :: usb :: DmPin , USB , PA11 , 0u8) ; pin_trait_impl ! (crate :: usb :: DpPin , USB , PA12 , 0u8) ; dma_trait_impl ! (crate :: i2c :: TxDma , I2C1 , { channel : DMA1_CH2 } , ()) ; dma_tra... 
  |                                                                         ^^^
  |                                                                         |
  |                                                                         unresolved import
  |                                                                         help: a similar path exists: `stm32_metapac::usb`

For more information about this error, try `rustc --explain E0433`.
error: could not compile `embassy-stm32` (lib) due to 2 previous errors
```

alexshirley, 2023-05-19:
I completely removed the `edition` section, and that seems to have done something

<details>
<summary>trace</summary>

```rust
error[E0425]: cannot find function `__basepri_r` in module `crate::asm::inline`
 --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri.rs:6:15
  |
6 |     call_asm!(__basepri_r() -> u8)
  |               ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_w` in module `crate::asm::inline`
  --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri.rs:22:19
   |
22 |         call_asm!(__basepri_w(basepri: u8));
   |                   ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_max` in module `crate::asm::inline`
  --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri_max.rs:19:19
   |
19 |         call_asm!(__basepri_max(basepri: u8));
   |                   ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__faultmask_r` in module `crate::asm::inline`
  --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\faultmask.rs:29:28
   |
29 |     let r: u32 = call_asm!(__faultmask_r() -> u32);
   |                            ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error: invalid register `r0`: unknown register
   --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\..\asm\inline.rs:197:24
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));
    |                        ^^^^^^^^^^^^^^

error: invalid register `r1`: unknown register
   --> C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\..\asm\inline.rs:197:40
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));
    |     
```
</details>

Dirbaio, 2023-05-19:
you need `edition=2021` because it defaults to [`resolver=2`](https://doc.rust-lang.org/cargo/reference/resolver.html#resolver-versions). Otherwise it's merging dependency features across deps and build-deps. For example, `invalid register r0: unknown register` fails because it's trying to build ARM ASM for x86, and x86 has no `r0`.

First, make sure you're using exactly the Rust nightly version that's supported by Embassy. Put [this file](https://github.com/embassy-rs/embassy/blob/master/rust-toolchain.toml) at the root of your project.

Second, if that still doesn't fix it: Can you change it to back when you were getting the USB-related error? and then please paste
- your full `Cargo.toml`
- your full `.cargo/config.toml`
- the output of `cargo tree --format '{p} {f}'`
- The output of `rustc --version`
- The output of `cargo --version`
- The output of `rustup show`

alexshirley, 2023-05-19:
Ok, I did put the rust-toolchain at the project root, and re-added the edition=2021, so now I'm back to the USB error

<details>
<summary>cargo.toml</summary>
[package]
authors = ["Alex Shirley"]
edition = "2021"
readme = "README.md"
name = "app"
version = "0.1.0"

[dependencies]
cortex-m = { version = "=0.7.6", features = ["inline-asm"] }
stm32-metapac = "7.0.0"
cortex-m-rt = "0.7.0"
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = "0.3"
cortex-m-semihosting = "0.3.3"
embassy-executor = {version = "0.2.0", path = "../embassy/embassy-executor", features = ["defmt", "integrated-timers"] }
embassy-time = {version = "0.1.0", path = "../embassy/embassy-time", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-usb = {version = "0.1.0", path = "../embassy/embassy-usb", features = ["defmt"]}
embassy-stm32 = {version = "0.1.0", path = "../embassy/embassy-stm32", features = ["defmt","stm32f070cb"]}
# embassy-executor = {version = "0.2.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "integrated-timers"] }
# embassy-time = {version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "defmt-timestamp-uptime"] }
# embassy-stm32 = {version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt","stm32f070cb"]}
static_cell = "1.0"

# this lets you use `cargo fix`!
[[bin]]
name = "app"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
</details>

<details>
<summary>.cargo/config.toml</summary>
[target.thumbv6m-none-eabi]
# uncomment this to make `cargo run` execute programs on QEMU
# runner = "qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel"
runner = 'probe-rs-cli run --chip STM32F070CBT6'

[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# uncomment ONE of these three option to make `cargo run` start a GDB session
# which option to pick depends on your system
runner = "arm-none-eabi-gdb -q -x openocd.gdb"
# runner = "gdb-multiarch -q -x openocd.gdb"
# runner = "gdb -q -x openocd.gdb"

rustflags = [
  # Previously, the linker arguments --nmagic and -Tlink.x were set here.
  # They are now set by build.rs instead. The linker argument can still
  # only be set here, if a custom linker is needed.

  # By default, the LLD linker is used, which is shipped with the Rust
  # toolchain. If you run into problems with LLD, you can switch to the
  # GNU linker by uncommenting this line:
  # "-C", "linker=arm-none-eabi-ld",

  # If you need to link to pre-compiled C libraries provided by a C toolchain
  # use GCC as the linker by uncommenting the three lines below:
  # "-C", "linker=arm-none-eabi-gcc",
  # "-C", "link-arg=-Wl,-Tlink.x",
  # "-C", "link-arg=-nostartfiles",
]
[... 254 lines cut ...]
installed toolchains
--------------------

stable-x86_64-pc-windows-msvc
nightly-2023-04-18-x86_64-pc-windows-msvc
nightly-x86_64-pc-windows-msvc (default)

installed targets for active toolchain
--------------------------------------

riscv32imac-unknown-none-elf
thumbv6m-none-eabi
thumbv7em-none-eabi
thumbv7em-none-eabihf
thumbv7m-none-eabi
thumbv8m.main-none-eabihf
wasm32-unknown-unknown
x86_64-pc-windows-msvc

active toolchain
----------------

nightly-2023-04-18-x86_64-pc-windows-msvc (overridden by 'C:\Users\Me\Documents\embed-rust\app\rust-toolchain.toml')
rustc 1.71.0-nightly (7908a1d65 2023-04-17)
</details>

Dirbaio, 2023-05-19:
Thanks for the detailed report! #1467 fixes it.

closed 2023-05-19 by Dirbaio

alexshirley, 2023-05-19:
Oh.... you're not going to believe this:

so I git pulled the changes to master ( #1467 )

and now I get some really strange stuff

<details>
<summary> Error log </summary>
```bash
   Compiling app v0.1.0 (C:\Users\Me\Documents\embed-rust\app)
warning: unused import: `embassy_executor::Spawner`
 --> src\main.rs:7:5
  |
7 | use embassy_executor::Spawner;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: unused import: `embassy_stm32 as stm32`
 --> src\main.rs:8:5
  |
8 | use embassy_stm32 as stm32;
  |     ^^^^^^^^^^^^^^^^^^^^^^

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\Me\\AppData\\Local\\Temp\\rustcZPlr1e\\symbols.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\app-d8edf52da0de2fc2.2tnkvuq8pqvz30q9.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\app-d8edf52da0de2fc2.2xeu7trzphn5pujx.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\app-d8edf52da0de2fc2.4zxatr14vrga0afj.rcgu.o" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\app-d8edf52da0de2fc2.5g5mqcwha8pw992w.rcgu.o" "--as-needed" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\debug\\deps" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\build\\app-8760c3e60e98fd68\\out" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-2a2c1cf939415100\\out" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-28b49e65b2596476\\out" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-semihosting-b682f0ff348d29b8\\out" "-L" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\build\\defmt-8af0000efb59e6aa\\out" "-L" "C:\\Users\\Me\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\stm32-metapac-7.0.0/src/chips/stm32f070cb" "-L" "C:\\Users\\Me\\.rustup\\toolchains\\nightly-2023-04-18-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libpanic_probe-c6df554a64873c57.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_stm32-ea6753ea160ccbf8.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libfutures-9428502ca2abc55e.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_futures-0a0d15d9932503ef.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembedded_storage-213280e2fa07a940.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_embedded_hal-f6af4498d8867005.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembedded_hal-66ba30afb3c348a4.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libstm32_metapac-751fde3e6cb74d9a.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_sync-54a8ae41eebb490a.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libheapless-88c14b24539c091d.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libstable_deref_trait-d54310192f76bc06.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libatomic_polyfill-54484fb83421d978.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libhash32-f7b226c63c61ad1f.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libbyteorder-9c6528e151555ef1.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_cortex_m-e0858a265c528920.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_hal_common-eda9a73457b15546.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libnum_traits-45dbc6f3ffdcc840.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_executor-68422b19606b884a.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libstatic_cell-13c32f9fbde85c9f.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libembassy_time-cce134b172b18234.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libdefmt-2d92ec9d74de4867.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libbitflags-a1dbec0c0a73ce9f.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libfutures_util-449120aa96e26f14.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libpin_project_lite-d4e4fc15f4524910.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\debug\\deps\\libfutures_sink-24ef459451752031.rlib" "C:\\Users\\Me\\Documents\\embed-rust\\app\\target\\thumbv6m-none-eabi\\deb [... cut ...] ic_polyfill-54484fb83421d978.rlib
          >>> referenced by lib.rs:197 (C:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.1\src\lib.rs:197)
          >>>               atomic_polyfill-c8d03d23c85fc39e.atomic_polyfill.27d7baf0-cgu.0.rcgu.o:(core::ptr::drop_in_place$LT$critical_section..with..Guard$GT$::h104296ca1181c667) in archive C:\Users\Me\Documents\embed-rust\app\target\thumbv6m-none-eabi\debug\deps\libatomic_polyfill-c8d03d23c85fc39e.rlib
```
<\details>

alexshirley, 2023-05-19:
Disregard. Solution was to add 
```
"critical-section-single-core"
```

```
[dependencies]
cortex-m = { version = "=0.7.6", features = ["inline-asm","critical-section-single-core"] }
```

commit f43d5784 (2023-05-19): stm32/usb: do not require embassy-time.

Fixes #1466
commit 990b5ffd (2023-05-19): Merge #1467

1467: stm32/usb: do not require embassy-time. r=Dirbaio a=Dirbaio

Fixes #1466

bors r+
commit ea5c07d4 (2023-05-19): Merge #1467

1467: stm32/usb: do not require embassy-time. r=Dirbaio a=Dirbaio

Fixes #1466

bors r+
PR #1467 (2023-05-19): stm32/usb: do not require embassy-time.
Fixes #1466

bors r+