embassy-rs/embassy #1470: embassy-rp on std environemt

elpiel, 2023-05-21:
Hello,
It would be nice to continue the support for running tests on other platforms.
For example running `cargo test --target x86_64-unknown-linux-gnu --no-default-features -F std` on an embassy application to run test on host machine.

There seems to be some invalid registers and right now it fails with:

```
error: invalid register `r0`: unknown register
   --> /home/elpiel/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4b30394/embassy-rp/src/flash.rs:675:13
    |
675 |             in("r0") &cmd as *const FlashCommand,
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: invalid register `r1`: unknown register
   --> /home/elpiel/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4b30394/embassy-rp/src/flash.rs:676:13
    |
676 |             in("r1") ptrs,
    |             ^^^^^^^^^^^^^

error: invalid register `r2`: unknown register
   --> /home/elpiel/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4b30394/embassy-rp/src/flash.rs:677:13
    |
677 |             out("r2") _,
    |             ^^^^^^^^^^^

error: invalid register `r3`: unknown register
   --> /home/elpiel/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4b30394/embassy-rp/src/flash.rs:678:13
    |
678 |             out("r3") _,
    |             ^^^^^^^^^^^

error: invalid register `r4`: unknown register
   --> /home/elpiel/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4b30394/embassy-rp/src/flash.rs:679:13
    |
679 |             out("r4") _,
    |             ^^^^^^^^^^^

error: could not compile `embassy-rp` (lib) due to 5 previous errors
```

ssidarous-aurora, 2024-08-21:
Anyone doing this one ? is it still valid ? I'm looking to get involved with Embassy and this looks like a perfect first issue.

sheref-sidarous, 2024-08-21:
Anyone doing this one ? is it still valid ? I'm looking to get involved with Embassy and this looks like a perfect first issue.

Dirbaio, 2024-08-21:
you can run tests in std if you don't enable the `rt` feature. that's how embassy-rp itself is [tested in ci](https://github.com/embassy-rs/embassy/blob/main/.github/ci/test.sh#L24).

it'll of course crash if you try to actually use any of the peripheral drivers, so don't. but at leas thaving the dep won't fail the build.

For more complex setups i'd recommend splitting your code in two crates, one embedded-specific that depends on embassy-rp, and another that's just the logic you want to test.

elpiel, 2024-10-18:
Thank you for the response. I agree with the approach

closed 2024-10-18 by elpiel