embassy-rs/embassy #1514: Why use pio spi?

overheat, 2023-05-30:


overheat, 2023-06-11:
https://github.com/overheat/embassy/blob/82d765689aa0f922d0b43e402eaa3d911b2d461d/cyw43-pio/src/lib.rs

I can only find some comments in the code. Is it related to `overclock`?

phire, 2023-09-10:
On the Pico W, the SPI pins are not hooked to the correct pins to use the RP2040's normal SPI peripheral. The Pico W PCB also uses resistors to multiples MOSI and MISO (and the cyw43's interrupt request) onto a single GPIO pin, so it's not even proper 4-wire SPI anymore. 
This design allows the Pico W to have the same external GPIO layout as the original Pico, at the cost of needing to use the PIO as a custom SPI implementation.

That previous version might have been for a different PCB that hooked the cyw43 to the proper pins for SPI?

9names, 2024-11-25:
@overheat is this question answered sufficiently? if so, would you mind closing the issue?

overheat, 2024-11-25:
sure.

closed 2024-11-25 by overheat