embassy-rs/embassy #1519: Disconnect wifi to save energy

lucasmerlin, 2023-05-30:
Hi! Is there any way to disconnect the wifi after the connection has been established?

Frostie314159, 2023-08-23:
Any updates on this?

phire, 2023-08-26:
The energy cost of reconnecting is somewhat high, so if you are planning to reconnect every few min, it would be more sane to just to put the wifi into the `SuperSave` power management mode and leave it idle. 

I'm not sure how long your polling period would have to be to actually save power by disconnecting. 

-----

It wouldn't be too hard for someone to implement disassociation. According to [the upstream driver](https://github.com/georgerobotics/cyw43-driver/blob/8ef38a6d32c54f850bff8f189bdca19ded33792a/src/cyw43_ctrl.c#L651-L654C1), it's just a matter of sending a single ioctl.

But, from what I can tell, you need to to fully de-initialize the wifi chip to get the lowest power consumption, which would require quite a bit of refactoring. Either the job of uploading the firmware needs to be moved inside the wifi runner, or embassy-executor needs to gain functionality to stop tasks and start them again later.

Frostie314159, 2023-08-26:
Prehaps we could make use of the target wake functionality, which was made available in wifi 6.

leftger, 2026-10-09:
This looks like it was addressed — cyw43 now exposes an explicit disconnect.

`cyw43::Control::leave()` exists (`cyw43/src/control.rs`), which is the "disconnect to save energy" operation this was asking for.

Reopen or comment if you meant something finer-grained (e.g. entering a lower-power radio state rather than disconnecting).

closed 2026-10-12 by xoviat