embassy-rs/embassy #1533: Using 3rd party driver crates with embassy

Rhvarrier, 2023-06-04:
Hello,

I'm considering porting an existing rust embedded project to embassy and would like to know if embassy has any support for [driver crates](https://github.com/rust-embedded/awesome-embedded-rust#driver-crates) or should I rewrite them.
From what I understand embassy doesn't seem to follow all of [embedded-hal](https://crates.io/crates/embedded-hal). If so is there any plan to port the existing drivers to be compatible with embassy ?

Dirbaio, 2023-06-04:
Embassy HALs implement most `embedded-hal` and `embedded-hal-async` traits, so most drivers should work as-is.

leftger, 2026-10-09:
This looks like it may already be answered by the book.

`docs/pages/layer_by_layer.adoc` covers "Implementing the embedded-hal traits to make peripherals useful in third party drivers", and `faq.adoc` has an `embassy-embedded-hal` entry. That is the intended route for using external driver crates with embassy.

Reopen or comment if your question was about something the docs don't cover.

closed 2026-10-09 by leftger