embassy-rs/embassy #1537: Pi Pico HID device

nxy7, 2023-06-05:
Hello, I'm trying to make foot pedal like one of elgato products using Pi Pico. It seems that there are no RP Pico HID device examples, and code i hack together doesn't seem to be working. I've repo https://github.com/nxy7/piPicoFootPedal and the program is stuck at 
https://github.com/nxy7/piPicoFootPedal/blob/6f4f69cbec2dc009057432537fc4e4a189717d78/src/main.rs#L74-L93
I've also noticed that the device never shows up when I use `lsusb` and after running `usb.run().await` my espectation would be that it should appear there even if the rest of the functionality would have some problems.

If someone more experienced could have a look at the code and tell me what mistake did I make or if HID devices are not supported for Pi Pico yet, that would be great.

Dirbaio, 2023-06-05:
HID should work, yes. It's all hardware-independent above the driver layer.

try `run()` instead of `run_until_suspend()`. The latter exits if the USB device is suspended, at which point it'll permanently stop working until reboot.

nxy7, 2023-06-05:
I've tried both, but the code never gets past `hid.ready().await;`. I've suspected that there might be some other issue with the program as `usb.run().await` is running in the background as it should, but the device never shows up on device list anyway.

Dirbaio, 2023-06-05:
I just tried your repo, and it does work. Commit 6f4f69cb, changing `run_until_suspend` to `run`, no other changes. It types "aaaaaaa"

```
[dirbaio@mars piPicoFootPedal]$ cargo run --release
...
2.571495 INFO  Device suspended, the Vbus current limit is 500µA (or 2.5mA for high-power devices with remote wakeup enabled).
└─ main::{impl#2}::suspended @ src/main.rs:172
2.571514 INFO  Device suspended, the Vbus current limit is 500µA (or 2.5mA for high-power devices with remote wakeup enabled).
└─ main::{impl#2}::suspended @ src/main.rs:172
2.571533 INFO  Device suspended, the Vbus current limit is 500µA (or 2.5mA for high-power devices with remote wakeup enabled).
└─ main::{impl#2}::suspended @ src/main.rs:172
2.571552 INFO  Device suspended, the Vbus current limit is 500µA (or 2.5mA for high-power devices with remote wakeup enabled).
└─ main::{impl#2}::suspended @ src/main.rs:172
2.579208 INFO  Bus reset, the Vbus current limit is 100mA
└─ main::{impl#2}::reset @ src/main.rs:151
2.702355 INFO  Bus reset, the Vbus current limit is 100mA
└─ main::{impl#2}::reset @ src/main.rs:151
2.828826 INFO  USB address set to: 31
└─ main::{impl#2}::addressed @ src/main.rs:156
2.848525 DEBUG SET_CONFIGURATION: configured
└─ embassy_usb::{impl#2}::handle_control_out @ embassy/embassy-usb/src/fmt.rs:125
2.848619 INFO  Device configured, it may now draw up to the configured current limit from Vbus.
└─ main::{impl#2}::configured @ src/main.rs:162
2.848681 INFO  hid ready
└─ main::____embassy_main_task::{async_fn#0}::{async_block#0} @ src/main.rs:76
2.849155 INFO  Set idle rate for None to 4294967295
└─ main::{impl#0}::set_idle_ms @ src/main.rs:117
4.848815 INFO  Done ()
└─ main::____embassy_main_task::{async_fn#0}::{async_block#0} @ src/main.rs:89
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa6.848877 INFO  Done ()
└─ main::____embassy_main_task::{async_fn#0}::{async_block#0} @ src/main.rs:89
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa8.848914 INFO  Done ()
└─ main::____embassy_main_task::{async_fn#0}::{async_block#0} @ src/main.rs:89

nxy7, 2023-06-05:
Well it definitely is not working on my PC, the fact that it works for you means it's somehow related to my setup and the logic is fine. Thank you for confirming that, I'll check few different cabled and maybe try it under different PC entirely. Thank you so much for confirming the code is fine :-)

nxy7, 2023-06-05:
Turns out the issue were my usb cables. I was kind of unlucky cause 3/5 of my cables wouldn't work with the Pico. Thanks for saving me hours looking at the code that's working @Dirbaio ^_^

closed 2023-06-05 by nxy7