embassy-rs/embassy #1542: Feature Request: USB Video Class implementation

ZanzyTHEbar, 2023-06-07:
Loving embassy and what ya'll are doing. Would be great if you offered UVC as a built-in class :) 

If this is not in the plan, maybe better docs on how to use the usb builder API to make a UVC class.

Dominaezzz, 2024-01-27:
I'm looking into building something for this but it looks like the USB stack here doesn't support isochronous endpoints yet which is required for this.

https://github.com/embassy-rs/embassy/blob/fb22b46ebb40c16e35c651c0dacf810126856927/embassy-usb-driver/src/lib.rs#L30-L31

Noxime, 2025-06-04:
Hi, I would be interested in UVC functionality as well. Has the situation with isochronous transfers improved?

Dominaezzz, 2025-06-04:
https://github.com/embassy-rs/embassy/pull/3314

leftger, 2026-11-07:
#7447 addresses this

closed 2026-11-07 by leftger