embassy-rs/embassy #1543: Problem when trying to connect with tcp socket on rarspberry pico w

teovillanueva, 2023-06-08:
Hello! First time opening an issue here. Im new to embedded systems and rust. Im trying to open a connection and im getting this error `ERROR panicked at 'adding a socket to a full SocketSet'`. Apparently this error comes from the `smoltcp` crate. I've been trying to debug this but I just cant figure it out. This is what Im doing:

```rs
let mut rx_buffer = [0; 4096];
let mut tx_buffer = [0; 4096];

let mut socket = TcpSocket::new(stack, &mut rx_buffer, &mut tx_buffer);

socket.set_timeout(Some(Duration::from_secs(10)));

let host_addr = embassy_net::Ipv4Address::from_str("192.168.1.3").unwrap();

if let Err(e) = socket.connect((host_addr, 1883)).await {
    warn!("connect error: {:?}", e);
}
```

Dirbaio, 2023-06-08:
When you create the network stack, there's this: `make_static!(StackResources::<2>::new()),`. That `2` there is the max number of sockets. Increase it. (Note it needs 1 extra socket for DHCP, so if you want 1 TCP socket then you need to set it to at least `2`)

teovillanueva, 2023-06-08:
That was it! Thank you so much!

closed 2023-06-08 by teovillanueva