embassy-rs/embassy #1547: embassy-stm32: make some dependencies optional

wiktorwieclaw, 2023-06-09:
* `embassy-executor` - From what I've understood, `embassy-stm32` is meant to be usable with other executors (e.g the one from `rtic`). The dependency is already optional for `embassy-nrf`.
* `embassy-net-driver` - well, not everybody needs a net driver

Are there any blockers? Just give me a hands up and I'll make a PR.

Dirbaio, 2023-06-09:
> embassy-executor - From what I've understood, embassy-stm32 is meant to be usable with other executors (e.g the one from rtic). The dependency is already optional for embassy-nrf.

You can still use it with other executors even if the dep is there. Also it doesn't consume flash space if you don't use it, the linker doesn't include it. I'm OK with removing executor reexports from HALs though (not even making them optional) since nobody's using them anyway.

> embassy-net-driver - well, not everybody needs a net driver

Same as above, it doesn't consume flash space if you don't use it. The only benefit of making it optional is compile time, and only for the initial build. And `embassy-net-driver` is a tiny crate with just the driver trait (it doesn't build the whole embassy-net/smoltcp), so the compile time impact is not much anyway.

It is an intentional design decision to *not* gate things with features if the only benefit is "reduce compile time a bit". It makes the crate unnecessarily harder to use, for little benefit.

wiktorwieclaw, 2023-06-09:
Thanks for the explanation.

We can preserve the ease of use by making the dependencies enabled by default, while giving the option to disable them to people that like to keep their dependency graph as small as possible.

If you don't feel it's worthwhile, feel free to close the issue. ;P

Dirbaio, 2023-06-09:
the trouble with default features is it only takes one lib that depends on `embassy-stm32` to forget `default-features = false`, and now all the features get enabled for you even if you do add `default-features = false`.

commit b51150d1 (2023-06-09): Remove executor dep+reexports from HALs.

Closes #1547
commit dc8e3442 (2023-06-09): Remove executor dep+reexports from HALs.

Closes #1547
closed 2023-06-09 by Dirbaio