embassy-rs/embassy #1554: I2C slave/peripheral mode support for rp2040

Ynng, 2023-06-12:
I2C slave mode is currently not supported on rp2040.

`rp-hal` have an implementation already, would it be straight forward to write 
a similar implementation based on theirs?
https://github.com/rp-rs/rp-hal/pull/162

CBJamo, 2023-06-16:
I do have some work[1] done on this front, but got pulled away by other priorities before I could complete it. I was more-or-less copying the interface of the nrf i2c device. The basic functionality works, but needs to be more thoroughly tested. I probably won't be able to get back to this till mid-July, maybe early August.

[1] https://github.com/CBJamo/embassy/tree/i2cdev

Ynng, 2023-06-16:
Thanks, https://github.com/CBJamo/embassy/tree/i2cdev is working well for me

CBJamo, 2024-03-26:
This has been merged into embassy-rp. I recommend using a recent git revision, as there are improvements that have not yet been released.

Dirbaio, 2024-03-26:
closing since this is done now.

closed 2024-03-26 by Dirbaio