embassy-rs/embassy #1564: Any guidance on the correct way to allow `cargo test` to work?

sureshjoshi, 2023-06-16:
I'm a bit stuck on something, and was hoping someone could help me out. This might be more of a generalized Rust question, but it applies to Embassy directly in my case.

I see the CI uses cargo test on many of the workspace projects, but whenever I run `cargo test` on my project I get `error[E0463]: can't find crate for `test``.

I'm assuming this is because `#![no_std]` and the various ARM targets (e.g. `thumbv7em-none-eabi`). 

So, is the test-centric guidance to create separate crates/libs for the code that can be tested without hardware, and pull those into the final hardware-specific binaries? Or should I be looking into no_std testing libraries? Or something else/in-between? 

The former method is what I've been doing forever in C/C++ embedded projects, but I wasn't sure if in Rust I just need a set of feature flags applied to automagically take care of std/no_std and tests. The former method is also how I think testing of embassy-sync and other crates appears to be working, since I don't see `no_std` there, but rather some config attributes pointing at `std`.

As an example, I'm trying to write unit tests for a simple 6 line pure function 32/64-bit agnostic, with no explicitly hardware/ARM-specific code. Just a little helper for something that will run on my NRF52840.

Thanks!

Dirbaio, 2023-06-16:
Testing a nostd crate is a bit hit or miss, yes...

You need to make the crate no_std only when not testing `#![cfg_attr(not(test), no_std)]`. This will still build embedded-only libs for x86, which might or might not work. You might have to adjust features for the deps, for example `embassy-nrf` itself needs `default-features=false` to get the `rt` feature disabled (this is how it's [tested in CI](https://github.com/embassy-rs/embassy/blob/main/.github/ci/test.sh#L24)). 

Alternately, separate the logic into a separate crate with no embedded-only deps and test just that, yes.

sureshjoshi, 2023-06-16:
Awesome, thanks for the quick response @Dirbaio !

From what I can tell, I kind of need to do both. 

I've split up my cross-platform code into a separate crate, got `cargo test` running - but to import it back into my NRF code, it requires `#![cfg_attr(not(test), no_std)]`

I'm sure I could also flip the logic at some point, but that doesn't really matter right now. 

Either way, I definitely think this separation of platform-specific code vs platform-agnostic code is a much "cleaner" approach.

---

For posterity, I've also been trying the `extern crate std` approach that I've read about in a rust thread, so the file is declared no_std, and then just the test module externs. This works in my very trivial tests so far.

```rust
#![no_std]

pub fn add(a: u8, b: u8) -> u8 {
    a + b
}

#[cfg(test)]
mod tests {
    use super::*;
    extern crate std;

    #[test]
    fn it_works() {
        _ = std::vec![1, 2, 3];
        let result = add(2, 2);
        assert_eq!(result, 4);
    }
}
```

closed 2023-06-16 by sureshjoshi