embassy-rs/embassy #1575: How create new project?

simeondmr, 2023-06-21:
Hello,

I'm new in Embassy...
How can I create a blank project for example for "stm32g0"?

k12ish, 2023-07-24:
I think this issue has some merit.

- Example `Cargo.toml` files would have to be modified to remove the relative imports:

https://github.com/embassy-rs/embassy/blob/8d50f8a3d3e5a50cd266ca45e221565e5982e744/examples/stm32g0/Cargo.toml#L7-L12

- The wiki is missing a section for creating a new project: 

https://github.com/embassy-rs/embassy/wiki/Getting-Started#create-a-project-using-embassy

Perhaps this issue could be tagged with "Documentation" and "Good First Issue"?

snakehand, 2023-08-08:
I tried changing the relative paths from an example to git paths, and got 77 errors like this:

154 |         let flush_res = bus.flush().await;
    |                                    -^^^^^
    |                                    ||
    |                                    |`impl ?Sized` is not a future
    |                                    help: remove the `.await`
    |
    = help: the trait `Future` is not implemented for `impl ?Sized`
    = note: impl ?Sized must be a future or must implement `IntoFuture` to be awaited
    = note: required for `impl ?Sized` to implement `IntoFuture`

Dirbaio, 2023-08-08:
@snakehand you're using a nightly version that has bugs. 

Use this exact version, that's the one supported by Embassy and known to be bug-free: https://github.com/embassy-rs/embassy/blob/main/rust-toolchain.toml#L4
easiest way is to put that `rust-toolchain.toml` file in your project.

k12ish, 2023-12-20:
Commenting for posterity -- The docs have a section describing [how to setup a new project](https://embassy.dev/book/dev/new_project.html)

[Permalink](https://github.com/embassy-rs/embassy/blob/d832d45c0ba5f2624a5f5c1e549e2d7fe8bd0e01/docs/modules/ROOT/pages/new_project.adoc)

lulf, 2023-12-21:
The documentation has definitely improved since this issue was raised, closing.

closed 2023-12-21 by lulf