embassy-rs/embassy #1579: Is the newer STM32U5 family such as the STM32U5A5ZJ supported?

jerabaul29, 2023-06-23:
I know that "older" stm32u5 processors are supported (I have a NUCLEO-U575ZI-Q https://www.st.com/en/evaluation-tools/nucleo-u575zi-q.html using the STM32U575ZIT6Q and I am able to run embassy programs on it just fine so far). Do you know if some of the "brand new" stm32u5 processors, such as the STM32U5A5ZJ used in the NUCLEO-U5A5ZJ-Q https://www.st.com/en/evaluation-tools/nucleo-u5a5zj-q.html (released Q2 2023) are supported too? I will buy one of these, I can also test / report how things go when I get it :) .

jerabaul29, 2023-06-23:
Does the fact that it is named here:

https://github.com/embassy-rs/embassy/blob/4dd48099bee305fc31e47a586f34d09c3ec02673/embassy-stm32/Cargo.toml#L1408

mean that it is supported? :)

Dirbaio, 2023-06-23:
STM32U5[78] are tested working on actual hardware. I later added STM32U5[9A] based on ST's XMLs but I haven't checked anything beyond "it compiles" (I haven't been able to get my hands on a U5[9A] board yet). They seem to be very similar just with added peripherals, so if we're lucky it'll Just Work. If we're not lucky it might require some small fixes :)

Dirbaio, 2023-10-04:
since my last comment embassy-stm32 has been tested working on nucleo-u5a5zj by both another person and me, it's up and running fine. There might still be missing bits of course, but the basics are there.