embassy-rs/embassy #1584: rx_chan.rx_buf() gets stuck when running nmap!

vDorst, 2023-06-25:
I am currently working on a new ethernet driver for the Singe Pair Ethernet chip ADIN1110.

I first implemented the driver for RTIC with SmolTCP.
But I hit the issue that SmolTCP is not async so I rewrote the driver to use Embassy and `net-driver-channel`.
Driver works, basic ping and tcp echo works, but when I running nmap to my "EVAL-ADIN1110EBZ¨" board I see that the driver gets stuck at getting a new buffer allocated.
I used the `embessy-net-w5500`as reference and used the `rp/tcp-server` example as demo/reference for my driver om a STM32L4S5 device.

Current `runner` code loops like this. 
Driver get stuct at ` mut frame = rx_chan.rx_buf().fuse() =>`.
Is this a bug or do I something wrong?

```
impl<'d, SPI: SpiDevice, INT: Wait, RST: OutputPin> Runner<'d, SPI, INT, RST> {
    pub async fn run(mut self) -> ! {
        loop {
            let (state_chan, mut rx_chan, mut tx_chan) = self.ch.split();

            // defmt::println!("Set MAC {}", MAC);
            // spe.set_mac_addr(&MAC).await.unwrap();

            #[cfg(feature = "defmt")]
            loop {
                defmt::println!("Waiting for interrupts");
                select_biased! {
                    _ = self.int.wait_for_low().fuse() => {
                        let mut status1_clr = 0x00;
                        let status0 = self.mac.read_reg(sr::STATUS0).await.unwrap();
                        let mut status1 = self.mac.read_reg(sr::STATUS1).await.unwrap();
                        if status1 & !0x1b != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPE CHIP STATUS 0:{:08x} 1:{:08x}", status0, status1);
                        }

                        while status1 & 0x10 != 0 {
                            defmt::println!("alloc account");
                            select_biased! {
                                mut frame = rx_chan.rx_buf().fuse() => {  // <-- Gets stuck here
                                    // defmt::println!("read packet from device");
                                    match self.mac.read_fifo(&mut frame).await {
                                        Ok(n) => {
                                            // #[cfg(feature = "defmt")]
                                            // defmt::println!(
                                            //     "RX: PACKET: [{}] {:02x}",
                                            //     n,
                                            //     &frame[0..n]
                                            // );
                                            // #[cfg(feature = "defmt")]
                                            // defmt::println!("from_wire: [{}]", n);
                                            rx_chan.rx_done(n);
                                            // #[cfg(feature = "defmt")]
                                            // defmt::println!("from_wire done");
                                        }
                                        Err(e) => match e {
                                            AdinError::PACKET_TOO_BIG => {
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Packet to big, DROP");
                                                self.mac.write_reg(sr::FIFO_CLR, 1).await.unwrap();
                                            }
                                            AdinError::Spi(_) =>{
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Spi error")
                                            },
                                            _ => {
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Error")
                                            },
                                        },
                                    }
                                }
                                _ = Timer::after(Duration::from_millis(100)).fuse() => {
                                    defmt::println!("Alloc error");
                                    break;
                                }
                            }
                            status1 = self.mac.read_reg(sr::STATUS1).await.unwrap();
                            defmt::println!("status: {:08x}", status1);
                        }

                        if status1 & 0x08 != 0 {
                            status1_clr |= 0x08;
                            #[cfg(feature = "defmt")]
                            defmt::println!("TX_DONE");
                        }

                        if status1 & 0x2 != 0 {
                            let link = status1 & 0x01 != 0;
                            self.is_link_up = link;
                            state_chan.set_link_state( if link {  LinkState::Up } else { LinkState::Down } );
                            #[cfg(feature = "defmt")]
                            defmt::println!("LINK Changed: Link {}", link);
                            status1_clr |= 0x02;
                        }

                        if status1 & 0x1000 != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPI TX_ECC_ERR error, CLEAR TX FIFO");
                            self.mac.write_reg(sr::FIFO_CLR, 2).await.unwrap();
                            status1_clr |= 0x1000;
                        }

                        if status1 & 0x400 != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPI SPI_ERR CRC error");
                            status1_clr |= 0x400;
                        }

                        if status0 & 0x80 != 0 {
                            let crsm_irq_st = self.mac
                                .read_cl45(MDIO_PHY_ADDR, RegsC45::DA1E::CRSM_IRQ_STATUS.into())
                                .await
                                .unwrap();

                            let phy_irq_st = self.mac
                                .read_cl45(MDIO_PHY_ADDR, RegsC45::DA1F::PHY_SYBSYS_IRQ_STATUS.into())
                                .await
                                .unwrap();
                            #[cfg(feature = "defmt")]
                            defmt::println!(
                                "SPE CHIP PHY CRSM_IRQ_STATUS {:04x} PHY_SUBSYS_IRQ_STATUS {:04x}",
                                crsm_irq_st,
                                phy_irq_st
                            );
                        }

                        if status0 & (1 << 11) != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPE CHIP PHY TX Frame CRC error");
                        }

                        // Clear status0
                        self.mac.write_reg(sr::STATUS0, 0xFFF).await.unwrap();
                        self.mac.write_reg(sr::STATUS1, status1_clr).await.unwrap();
                    }
                    mut packet = tx_chan.tx_buf().fuse() => {
                        self.mac.write_fifo(&mut packet).await.unwrap();
                        tx_chan.tx_done();
                    },
                }
            }
        }
    }
```

Dirbaio, 2023-06-25:
Yeah, buffer management in the channel driver is a bit tricky :sweat: 

What's happening is a deadlock: The embassy-net task is stalled because the tx queue is full, therefore it's not processing packets in the rx queue. Your task is stalled because the rx queue is full, therefore it's not processing packets in the tx queue.

The fix is to make sure to always service the tx queue while you're waiting for space to become available in the rx queue. For example, select on either "tx_chan.tx_buf() available" or "INT is low AND rx_chan.rx_buf() available". This is why [this select](https://github.com/embassy-rs/embassy/blob/main/embassy-net-w5500/src/lib.rs#L53-L59) is written like that.

Dirbaio, 2023-06-25:
BTW, ADIN1110 looks very interesting! Any chance you could consider contributing the driver? :eyes:

vDorst, 2023-06-26:
Thanks for the information.

Adding a extra select_biased!() in the RX path that prefers to handle the rx_chan but handles also tx_chan if rx_chan is blocking, solves it. Now is keeps running and reconfers from the nmap flood.

Code now looks like this
```
impl<'d, SPI: SpiDevice, INT: Wait, RST: OutputPin> Runner<'d, SPI, INT, RST> {
    pub async fn run(mut self) -> ! {
        loop {
            let (state_chan, mut rx_chan, mut tx_chan) = self.ch.split();

            loop {
                #[cfg(feature = "defmt")]
                defmt::println!("Waiting for interrupts");
                select_biased! {
                    _ = self.int.wait_for_low().fuse() => {
                        let mut status1_clr = 0x00;
                        let mut status1 = self.mac.read_reg(sr::STATUS1).await.unwrap();

                        while status1 & 0x10 != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("alloc RX packet buffer");
                            select_biased! {
                                // Handle frames that needs to transmit from the wire.
                                // Note: rx_chan.rx_buf() channel don´t accept new request
                                //       when the tx_chan is full. So these will be handled
                                //       automaticly.
                                mut frame = rx_chan.rx_buf().fuse() => {
                                    match self.mac.read_fifo(&mut frame).await {
                                        Ok(n) => {
                                            rx_chan.rx_done(n);
                                        }
                                        Err(e) => match e {
                                            AdinError::PACKET_TOO_BIG => {
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Packet to big, DROP");
                                                self.mac.write_reg(sr::FIFO_CLR, 1).await.unwrap();
                                            }
                                            AdinError::Spi(_) =>{
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Spi error")
                                            },
                                            _ => {
                                                #[cfg(feature = "defmt")]
                                                defmt::println!("RX Error")
                                            },
                                        },
                                    }
                                }
                                // Handle frames that needs to transmit to the wire.
                                mut packet = tx_chan.tx_buf().fuse() => {
                                    self.mac.write_fifo(&mut packet).await.unwrap();
                                    tx_chan.tx_done();
                                },
                                // _ = Timer::after(Duration::from_millis(100)).fuse() => {
                                //     #[cfg(feature = "defmt")]
                                //     defmt::println!("RX alloc error");
                                //     break;
                                // }
                            }
                            status1 = self.mac.read_reg(sr::STATUS1).await.unwrap();
                            defmt::println!("status: {:08x}", status1);
                        }

                        let status0 = self.mac.read_reg(sr::STATUS0).await.unwrap();
                        if status1 & !0x1b != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPE CHIP STATUS 0:{:08x} 1:{:08x}", status0, status1);
                        }

[... 45 lines cut ...]

                        if status0 & (1 << 11) != 0 {
                            #[cfg(feature = "defmt")]
                            defmt::println!("SPE CHIP PHY TX Frame CRC error");
                        }

                        // Clear status0
                        self.mac.write_reg(sr::STATUS0, 0xFFF).await.unwrap();
                        self.mac.write_reg(sr::STATUS1, status1_clr).await.unwrap();
                    }
                    // Handle frames that needs to transmit to the wire.
                    mut packet = tx_chan.tx_buf().fuse() => {
                        self.mac.write_fifo(&mut packet).await.unwrap();
                        tx_chan.tx_done();
                    },
                }
            }
        }
    }
}
````

I already had in mind to contribute this driver, still figuring out how it all works.

BTW: should it prefer the `embassy::select` or the rust futures `select_biases`?

Dirbaio, 2023-06-26:
> Adding a extra select_biased!() in the RX path that prefers to handle the rx_chan but handles also tx_chan if rx_chan is blocking, solves it.

Nice!! To avoid dupication of the TX maybe you could have 3 branches in the top-level select: handle irqs, handle rx, handle tx. Handling irqs would set some flag that would let the rx branch know it has to do stuff. 

```rust
let mut rx_pending = false;
select3(
   self.int.wait_for_low(),
   tx_chan.tx_buf(),
   async {
       if !rx_pending { futures::pending::<()>().await; } // if no rx pending, wait forever so this branch never gets picked.
       rx_chan.rx_buf().await
   }
)
// irq branch sets rx_pending = true
// rx branch resets rx_pending = false
```

This only works if the chip lets you "clear" the "packet rx'd" interrupt  of course... If it's "sticky" as in "it is always set until you actually pop the packet off the FIFO" then it won't work, the irq will be permanently deasserted...

> I already had in mind to contribute this driver, still figuring out how it all works.

:heart: 

> BTW: should it prefer the embassy::select or the rust futures select_biases?

`embassy_futures` `select` should be a bit lighter since it doesn't require fusing. It's unlikely to make a difference in practice.

vDorst, 2023-06-26:
You can't clear the "has packet flag", just by reading the fifo buffer it will clear this flag.

I put my current state here of the driver here: https://github.com/embassy-rs/embassy/commit/3c7908704a588504953a083802a7c30cbad0abcc

I shall also make an example when the code is more ready.
I also struggeling a bit how to define the registers.
I like the `tock-registers`  crate idea but I sure how to apply it and if this is the right way.

closed 2023-08-17 by vDorst