embassy-rs/embassy #1596: `embassy-usb` APIs are very clunky

kaylynn234, 2023-06-29:
Hi there!

I've recently been working on implementing some device-agnostic keyboard firmware, and I've found the functionality offered by `embassy-usb` to be really, *really* clunky, so I wanted to open an issue to mention my frustrations and hopefully start a discussion on making some improvements.

My specific gripes mostly boil down to the way that the `Builder` type works - it's really not intuitive, and I've found implementing [USB DFU](https://www.usb.org/sites/default/files/DFU_1.1.pdf) pretty frustrating since the `Builder` APIs make writing descriptors really awkward. I've had a look around at the rest of the embedded USB ecosystem, and I think it would be worth considering implementing an API similar to that of `usb-device`, especially with regards to its `DescriptorWriter` type, which seems significantly easier to navigate.

(Weirdly enough, parts of the `DescriptorWriter` API seem to be copy-pasted into `embassy-usb`'s source code, and just not exposed publicly)

I think needing to provide references to multiple buffers when initializing a device is also pretty irritating, but I understand that there probably isn't much in the way of a better option for that specific case.

Anyways, I digress. Because of the sync-async coloring, it's probably not really feasible to implement some sort of "bridge" on top of `usb-device`, but in a perfect world I think that's probably what I'd want to see happen; there are already a lot of USB class/peripheral implementations on top of `usb-device`. which is a big help for projects that don't want to reinvent the wheel.

If this has been brought up before, if there are already plans for improvements, or if I'm just missing something, please let me know. Thank you!

Dirbaio, 2023-06-29:
First of all, embassy *did* use `usb-device` until early 2022. It would wrap `usb-device`, `usbd-serial` into an async API. This setup had a lot of issues due to the fact that `usb-device` polling needs `&mut` on all the classes: it'd force the USB task to own all classes and share them with a giant Mutex with user tasks. This'd result in complicated type signatures, and not-ideal perf/code size. `usb-device` drivers also don't handle the interrupts, leaving it up to the user, which required more heavy lifting in the async wrapper.

`embassy-usb` solved all these issues (it allows any task to own endpoints and read/write to them lock-free. No giant mutex needed, better perf). Moving async to the core driver allows it to handle its interrupts, abstracting them away from higher layers. It also allows using async when implementing classes, which helps a lot vs having to write the classes manual-polling-state-machine-style and then wrapping them with async externally.

(Side note, there's been efforts to fix some of these in `usb-device` itself. See the `endpoint-trait` branch in their repo. They've stalled though. `embassy-usb` is essentially `usb-device` with the `endpoint-trait` changes and written with `async` instead of manual state machines)

So no, the decision to "reinvent the wheel" was not made lightly. :) The stack is in much better shape now. The tradeoff is you can't use usb-device class impls as-is.

Second: about the descriptor writer:

It's designed this way to enforce at compile-time that descriptors are correct. For example, you can't write an endpoint descriptor if you haven't written an interface descriptor before that the endpoint belongs to. These things trigger runtime errors or malformed descriptors with the `usb-device` API, while they won't compile at all with the `embassy-usb` API.

You mention "clunky", "not intutiive", "awkward". These are subjective attributes, and not actionable to improve the API. Can you provide actionable insights on why you find them clunky? For example, "I'm trying to do X and it's not possible because Y is missing", "docs are missing for how to do Y", "doing Z has 6 steps which are hard to figure out, if we added $METHOD it would be 1 step with no loss of type-safety".

kaylynn234, 2023-06-29:
Thanks for your response! Regarding `usb-device`, that makes a lot of sense - I figured it wasn't really feasible, but didn't know the project actually had used it previously :)

One really simple thorn in the current API that tripped me up at first was the `builder.function(...)` -> `function_builder.interface()` step. For the common case it might help to have a top-level `builder.interface(...)` shortcut that takes the same arguments as `builder.function(...)` & calls `interface()` implicitly. In my case, this is the first time I've implemented a protocol on top of USB, and I've been following from the USB DFU spec - the way things were laid out, it almost seemed like `builder.function(...)` "wasn't the right thing to do" so to speak, since what I *wanted* to write was an interface.

I've also found it difficult to reuse code between i.e the runtime and upgrade DFU modes, for example? The crux of that is needing to keep references to different types depending on what I'd like to do next. The builder actually doing things *as it goes* rather than taking in a type that represents a *description* of the operation makes structuring/composing things more difficult too, I think.

As an aside:

I'm unsure whether it would be practical (whether for implementation or binary size reasons) but I almost wonder if something vaguely reminiscent of `clap`'s builder-style API would be reasonable here; some sort of system with a set of easily-constructable builder types that are "lazy" and don't commit configuration until later. Similarly to clap's approach, you could potentially manage type safety and descriptor validity in that scenario by limiting where those builder types can be passed.

The main issue with an approach like that is that you then need to iterate through your description of configuration and actually interpret it; allocating interface numbers, endpoints, writing descriptors, and so forth. Since the user might want to use some of that information later - particularly the interface numbers! - you would need to return a meaningful result that somehow encapsulates that information. It does seem doable though, I think?

Dirbaio, 2023-06-29:
Interfaces belong to functions. That's the way USB works. The `usb-device` API looks simpler because it simply doesn't support multiple functions. It's not "simpler", it's missing features.

The alternative builder API you propose (write everything as Rust structs, then "serialize" everything to binary descriptors when finishing building has already been tried, and results in worse code size.

kaylynn234, 2023-06-29:
Right - I figured as much. That's unfortunate and I suppose there isn't much that can be done in that case. I really appreciate the clarification and apologize for the bother - please forgive my naivety! I might open a documentation PR at some point that reiterates what's been said here w.r.t API design so that it's laid out more clearly, but other than that there probably isn't really much actionable here, like you said. Thank you for your time!

closed 2023-06-29 by kaylynn234