embassy-rs/embassy #1604: Feature Request: DMA/BDMA/RingBuf: read_exact (or similar)

alexferro, 2023-07-01:
I'm working on a project with circular buffering, but not using the USART (it's audio) and the DMA RingBuf implementation is a fairly decent fit to my needs, but I'd rather work with blocks of data, rather than sample by sample. Thanks to the wake behavior of the DMA interrupt, if I successfully get into the Pending state, the next time my task gets woken is when I will be able to read a full block of data. However, given data processing time, by the time I am back to waiting on the DMA, I've usuaulyl gotten in at least one sample of data, and as far as I can tell, there's no longer a way (after #1428 was merged) to determine if read with a given size will succeed or not without trying it and getting a partial read. This means I have a slightly ugly loop through the future to do a partial read and then enter pending until I know that my actual data is all ready.

However, read_exact, has a few pitfalls that read does not. Specifically it is likely to misbehave (I don't think unsafely) is such a way that you could effectively end up hanging or overruning if the buffer to read_exact isn't a pure fraction of the internal ring buffer size.

I've included my working (this has been modified slightly to be standalone, not part of my project, so it's missing type annotations properly, since they came from elsewhere in my codebase) code that does this as part of an async fn.  This was pieced together from the usart_ringbuf driver. 

For me, this is not a high priority, as I have this workaround, even if I feel it's a bit ugly. I've quite enjoyed using embassy otherwise, and I was expecting this portion to be ugly since I saw #702 and agree that the general case is hard to make safe.

```rust
pub struct MyDmaRingBuffer<DMA_CH, W> {
    ring: RingBuffer<'static, DMA_CH, W>,
}

impl MyDmaRingBuffer {
    pub async fn read_exact(&mut self, buffer: &mut [W]) {
        use core::future::poll_fn;
        use core::sync::atomic::{compiler_fence, Ordering};
        use core::task::Poll;

        let mut read_data = 0;
        let buffer_len = buffer.len();

        let dma = poll_fn(|cx| {
            self.ring.set_waker(cx.waker());

            compiler_fence(Ordering::SeqCst);

            if let Ok((len, _remaining)) = self.ring.read(&mut buffer[read_data..buffer_len]) {
                read_data += len;
                if read_data == buffer_len {
                    Poll::Ready(())
                } else {
                    Poll::Pending
                }
            } else {
                defmt::panic!("SAI DMA RingBuffer Read Overrun");
            }
        });
        dma.await
    }
}
```

xoviat, 2023-07-08:
> there's no longer a way (after https://github.com/embassy-rs/embassy/pull/1428 was merged) to determine if read with a given size will succeed or not without trying it and getting a partial read.

Yes, because the result can change between checking whether it will succeed and reading the data. This is simply protecting people from writing a race.

A `read_exact` helper method is likely an acceptable change (I can't see a problem with it at the moment); you can submit a pr if you want.

alexferro, 2023-07-16:
> Yes, because the result can change between checking whether it will succeed and reading the data. This is simply protecting people from writing a race.

Fair enough. 

I'll see if I can reason out any pitfalls of read_exact that would apply as a generic helper function so that any documentation I write will be accurate and then create a PR against an up to date version of embassy in the near future. It looks like I was able to restructure my code to just explicitly return a Future instead of implicitly doing so as an async fn, so that will match the rest of the logic better.

alexferro, 2023-08-14:
My apologies for not closing this when the PR got merged. I just noticed it was still open.

closed 2023-08-14 by alexferro