embassy-rs/embassy #1605: Feature Idea: DMA/BDMA/RingBuf: read_with_map for avoiding extra copies

alexferro, 2023-07-01:
I'm using the DMA RingBuf in my audio processing application, and while writing my code around it, I was thinking about how to avoid an extra copy of the data. It hit me that if the API supported some way to do the equivalent of iterator::map on the read data, I could fold in the u32 to f32 conversion (or anything really) my logic does inside the read loop. 

Based on looking at the code, I'm not sure if it'd be easy or performant to implement this, since it'd disrupt `copy_to` in the code base, and it may be superseded by whatever solution #702 gains in the future. It's also a specialized use case, for a peripheral that is less common, and I cannot think of any applicability to the usart driver that spawned the existence of the ring buffer.
But I thought I'd get it out there for the folks with more embedded Rust experience to think about.

tyler-gilbert, 2023-09-10:
@alexferro, I have done a lot of audio streaming on the STM32 (in C). I usually use the half complete/complete interrupts with a ring buffer to read/write the data. I created #1879 to track the issue. Do you think that approach would work for your use case? Are you using I2S/SAI or ADC/DAC?

alexferro, 2023-09-10:
@tyler-gilbert See my comment on your issue you opened.