embassy-rs/embassy #1618: Possibly trivial question(s) about spawning tasks

sureshjoshi, 2023-07-04:
Is there any particular "cost" to spawning tasks on the fly? As in, should all tasks be spawned ahead-of-time, and then left in an awaiting state on a signal/channel?

Or, is it a valid pattern to spawn-on-demand?

I guess the thing I don't fully understand is whether the task macro + spawner do any sort of embedded "magic" that I need to be on the lookout for - or whether it's just as straight-forward as I think it might be, which is basically just putting a pre-allocated task onto the queue of tasks the executor needs to keep an eye out for.

That was the impression the docs give me (https://embassy.dev/book/dev/runtime.html). I read it as `#[embassy_executor::main]` is just a regular old task that happens to be first + comes with a spawner.

But what I was less clear on was whether it's fair game to use the spawner from the non-main tasks, to spawn new tasks? 

For example: I have a BLE scanning task, and when I find a specific device, can I spawn that off into a separate connection task that lives for the duration of the connection - and then ... dies? Gets recycled? Or do I need to ensure that connection task is long-lived so that I can safely re-use it?

Another example, I have an ANT+ scanning task, and then I might have up to 3 ANT+ devices connected at once - so I was going to create a task with pool_size=3, but can I spawn and let them finish, and then re-spawn? Or should I keep a long-lived loop and handle them with a channel?

Dirbaio, 2023-07-04:
> 's just as straight-forward as I think it might be, which is basically just putting a statically allocated task onto the queue of tasks the executor needs to keep an eye out for.

yes, it's exactly that.

> But what I was less clear on was whether it's fair game to use the spawner from the non-main tasks, to spawn new tasks?

yes, that's totally fine.

> For example: I have a BLE scanning task, and when I find a specific device, can I spawn that off into a separate connection task that lives for the duration of the connection - and then ... dies? [...]

Yes, that's exactly how I'm handling BLE connections. On receive connection, spawn a task. When the connection dies, just let the task exit. When a task exits, it's marked as free in the pool, so next spawns can reuse it.

sureshjoshi, 2023-07-04:
Awesome thanks! Great to know it functionally works more or less as I thought it does :) 

> When a task exits, it's marked as free in the pool, so next spawns can reuse it.

This was the piece I was trying to dig through the code to find. I assumed it must have been how everything worked, but hadn't yet ran into the code that does it.

closed 2023-07-04 by sureshjoshi