embassy-rs/embassy #1625: nRF: Selecting power state

trembel, 2023-07-06:
Hi,

Is there a way to select the power state explicitly, e.g. turning RAM retention off and only waking up on interrupts?
I basically want to use nrf-softdevice crate, and as far as I understood it needs to be coupled with embassy and thus the async executor - or is it possible to use it with bare-metal interrupt management?

Dirbaio, 2023-07-06:
embassy-nrf + nrf-softdevice support "System ON sleep" out of the box. The system sleeps with WFI  when no task has work to do, resulting in current consumption of 3uA. Hopefully that's enough for your application. You can build devices with 1-2 year battery life with bluetooth on.

System OFF sleep is not supported yet. That one is very different though, it's not "wake up on interrupt", it can only wake up on GPIO and a few other very restricted things. And on wakeup it boots "from scratch" as if you had powered it on for the first time.

RAM retention is only useful for System OFF, it makes little difference for System ON sleep.

Dirbaio, 2023-07-06:
nrf-softdevice is designed for async, so you need the executor for that. For other stuff (say uart, spi, i2c..) you can of course still manually handle interrupts and use the PAC directly.

trembel, 2023-07-17:
> embassy-nrf + nrf-softdevice support "System ON sleep" out of the box. The system sleeps with WFI when no task has work to do, resulting in current consumption of 3uA. Hopefully that's enough for [...]

I know the nRF52 very well, and am starting to use Rust on it recently. For one of my applications I need the current draw to be <1.5uA, so at least I need to turn of RAM retention if in System ON, or else use System OFF.

The idea is to advertise over BLE only after a GPIO event coming from an accelerometer happens, therefore this should be possible. I will try to get it work with the PAC; At least, when RAM retention is off this shouldn't really interfere with the executor.

closed 2025-01-10 by Dirbaio