embassy-rs/embassy #1646: embassy_net::Stack::run thrashes

wyager, 2023-07-12:
Calling `stack.run()` thrashes the CPU (including preventing lower-priority tasks from running on rtic).

I added some logging here to test it on my chip - you can look at the diff from main to see what logging I added.

https://github.com/wyager/embassy/tree/wyager-ethernet-thrashing

This function gets run thousands of times a second, starving other stuff on the system: https://github.com/embassy-rs/embassy/blob/main/embassy-net/src/lib.rs#L666

@Dirbaio thought it might be this TODO https://github.com/embassy-rs/embassy/blob/82d765689aa0f922d0b43e402eaa3d911b2d461d/embassy-stm32/src/eth/mod.rs#L84

Dirbaio, 2023-07-13:
[relevant discussion](https://matrix.to/#/!YoLPkieCYHGzdjUhOK:matrix.org/$u1VXiP5KoNszPb4MHlk6-PqqOwZ9OOFSd_AXR-y7MjA?via=matrix.org&via=tchncs.de&via=grusbv.com)

so there's no way to get an irq on link change (in the general case. You could if the PHY has an irq output and you wire it up to an EXTI pin, but most boards don't.)

So the fix is to somehow do that polling in a background task, at some reasonable frequency (say, once every second), instead of hot-polling like we do now.

xoviat, 2023-07-15:
@wyager #1653 likely resolves the issue. however, we need a minimal reproducer with multiprio eth to confirm.

closed 2023-07-15 by Dirbaio

wyager, 2023-07-26:
Confirmed this is fixed. Lower-priority threads now run no problem @xoviat

PR #1653 (2023-07-15): stm32/eth: solve busy-loop polling
fixes #1646