embassy-rs/embassy #1651: Embassy-stm31f1- on bluepill - flash error

sundaritisthatis, 2023-07-14:
Finished release [optimized + debuginfo] target(s) in 0.07s
     Running `probe-run --chip stm32f103c6 target/thumbv7m-none-eabi/release/blinky`
(HOST) INFO  flashing program (15 pages / 15.00 KiB)
Error: Error while flashing

Caused by:
    The execution of 'program_page' failed with code 1. This might indicate a problem with the flash algorithm.

Dirbaio, 2023-07-14:
Check you have the right chip in Cargo.toml for embassy-stm32. Check that your `memory.x` is correct, in particular that flash is not too big. 

If it still doesn't work, please file an issue for probe-rs.

stappersg, 2023-08-13:
@sundaritisthatis  do known that open issues consume human energy.

andresv, 2023-10-10:
Can be closed.

closed 2023-10-10 by Dirbaio