embassy-rs/embassy #1657: Sending raw IEEE 802.11 frames.

Frostie314159, 2023-07-15:
Is it possible to send raw IEEE 802.11 frames with the cyw43 driver?

xoviat, 2023-07-16:
Yes and no. You can operate on the raw net driver but if you do that, you won't be able to use smoltcp

Dirbaio, 2023-07-16:
That's raw Ethernet frames though, not 802.11. I don't know if the hardware is capable of that, and if it is it is not implemented. Pull requests welcome :)

Frostie314159, 2023-07-16:
I would guess the hardware is capable of that, since other chips like the esp32 also have it. On that note, why are there references to the awdl protocol in the events for the cyw43 driver? As far as I'm aware only Broadcom chips have that.

Dirbaio, 2026-08-29:
embassy-net now uses xarxa instead of smoltcp #6860 which has support for "ethernet raw sockets" which allow sending/receiving full raw ethernet frames. 

There's still no support on cyw43 for sending raw 802.11 frames (actual raw wifi frames, LOWER level than ethernet). I don't know if the hardware is capable of doing it without wifi chip firmware hacking, so i'd say it's out of scope.

closed 2026-08-29 by Dirbaio