embassy-rs/embassy #1682: Change SPI frequency or release

little-arhat, 2023-07-23:
mpu9250 sensor only allows 1mhz SPI freq for configuration but can be used with 20mhz for data reads [0]. `mpu9250` crates provides ability to reconfigure SPI after initialization, example usage:
```rust
    let mut mpu = Mpu9250::imu_with_reinit(
        spi,
        ncs,
        &mut delay,
        &mut mpu9250::MpuConfig::imu(),
        |spi, ncs| {
            let (dev_spi, (scl, miso, mosi)) = spi.free();
            let new_spi = dev_spi.spi((scl, miso, mosi), mpu9250::MODE, 20.mhz(), clocks);
            Some((new_spi, ncs))
        },
    ).unwrap();
```

`Spi` from `embassy-stm32` doesn't allow releasing pins so above method of reconfiguration is not possible. `.reconfigure` method (and `spi::Config`) do not have frequency/baud_rate field available, so it seems there is no way to change SPI settings after it was created. Am I missing something? Would it be possible to implement either smth similar to `.free` or extend `Config` with frequency? 

[0] https://github.com/copterust/mpu9250/issues/6#issue-416483409

little-arhat, 2023-07-26:
proposal is to move freq to config (similarly to rpi hal) and deprecate current freq parameter.

xoviat, 2023-07-28:
PRs welcome

closed 2023-07-30 by xoviat