embassy-rs/embassy #1691: embassy/examples/rp blinky example fail on MacOS

overheat, 2023-07-26:
```
➜ cargo run --bin blinky
    Updating git repository `https://github.com/embassy-rs/lora-phy`
    Updating crates.io index
   Compiling proc-macro2 v1.0.66
   Compiling unicode-ident v1.0.11
   Compiling version_check v0.9.4
   Compiling syn v1.0.109
   Compiling defmt v0.3.5
   Compiling thiserror v1.0.44
   Compiling defmt-macros v0.3.6
   Compiling bitflags v1.3.2
error[E0463]: can't find crate for `core`
  |
  = note: the `thumbv6m-none-eabi` target may not be installed
  = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

error[E0463]: can't find crate for `compiler_builtins`

error[E0463]: can't find crate for `core`
   --> /Users/aaron/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bitflags-1.3.2/src/lib.rs:282:1
    |
282 | pub extern crate core as _core;
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate
    |
    = note: the `thumbv6m-none-eabi` target may not be installed
    = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `bitflags` (lib) due to 3 previous errors
warning: build failed, waiting for other jobs to finish...
```

Dirbaio, 2023-07-26:
the error message is literally telling you how to fix it:

>  = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

closed 2023-07-26 by Dirbaio

overheat, 2023-07-26:
Yes. The root cause is that by accedently install rust through brew.
Just `brew remove rust` and fix it.

CamelShoko, 2024-07-22:
`brew remove rust`

also works for me 

`  │  ~/rust-workspace/embassy/examples/nrf52840 │   main  cargo build --release --bin blinky                                                                                                                                                                                                        101 ✘ │ 15:37:42 
   Compiling bitflags v1.3.2
   Compiling critical-section v1.1.2
   Compiling nb v1.1.0
   Compiling byteorder v1.5.0
   Compiling stable_deref_trait v1.2.0
   Compiling cfg-if v1.0.0
   Compiling void v1.0.2
   Compiling proc-macro2 v1.0.86
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling semver v0.9.0
   Compiling futures-core v0.3.30
   Compiling pin-utils v0.1.0
   Compiling embassy-time-driver v0.1.0 (/Users/zhangxiang/rust-workspace/embassy/embassy-time-driver)
   Compiling futures-task v0.3.30
   Compiling pin-project-lite v0.2.14
error[E0463]: can't find crate for `core`
  |
  = note: the `thumbv7em-none-eabi` target may not be installed
  = help: consider downloading the target with `rustup target add thumbv7em-none-eabi`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `critical-section` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `bitflags` (lib) due to 1 previous error
error: could not compile `cfg-if` (lib) due to 1 previous error
error: could not compile `nb` (lib) due to 1 previous error
error: could not compile `void` (lib) due to 1 previous error
error: could not compile `byteorder` (lib) due to 1 previous error
error: could not compile `stable_deref_trait` (lib) due to 1 previous error
error: could not compile `futures-core` (lib) due to 1 previous error
error: could not compile `pin-utils` (lib) due to 1 previous error
error: could not compile `futures-task` (lib) due to 1 previous error
error: could not compile `pin-project-lite` (lib) due to 1 previous error`