embassy-rs/embassy #1711: LSE requires setting PWREN bit on STM32L4

bartekkowalski, 2023-07-29:
According to the reference manual RM0394, section 5.1.4, Backup domain Access: 
To make changes to the RTC registers, you first need to enable the power interface clock by setting the PWREN bits.

This is missing from the current code and the program will hang on `while !RCC.csr().read().lsirdy() {}` every time on my STM32L431.

Adding the line to set the bit fixes the issue:

```rust
RtcClockSource::LSE32 => {
    // 1. Unlock the backup domain
    RCC.apb1enr1().modify(|w| w.set_pwren(true));
    PWR.cr1().modify(|w| w.set_dbp(true));

    // 2. Setup the LSE
    RCC.bdcr().modify(|w| {
        // Enable LSE
        w.set_lseon(true);
        // Max drive strength
        // TODO: should probably be settable
        w.set_lsedrv(Lsedrv::HIGH);
    });

    // Wait until LSE is running
    while !RCC.bdcr().read().lserdy() {}
}
```

Dirbaio, 2023-07-30:
Can you send a PR?

bartekkowalski, 2023-07-30:
> Can you send a PR?

No worries, I'll put something together tomorrow.

MabezDev, 2023-07-31:
FYI I made a similar change for the F2 here: https://github.com/embassy-rs/embassy/pull/1721

bartekkowalski, 2023-08-01:
I've created the PR here: #1730

Let me know if you would like me to fix anything!

closed 2023-08-01 by xoviat