embassy-rs/embassy #1735: Unable to communicate via I2C1 on rp

felipebalbi, 2023-08-02:
Hi,

I slightly modified the example i2c_async to talk to a tmp117 instead. The device is definitely wired up correctly, but I never get an ACK from tmp117:

```
1.002804 ERROR panicked at 'called `Result::unwrap()` on an `Err` value: Abort(NoAcknowledge)', src\main.rs:96:60
└─ C:\...\src\index.crates.io-6f17d22bba15001f\panic-probe-0.3.1\src\lib.rs:104
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
Frame 0: HardFault @ 0x10004f8e
```

Here's the code for reference:

```
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_rp::init(Default::default());

    // Communicate with tmp117
    info!("Setting up I2C");
    let instance = p.I2C1;
    let sda = p.PIN_14;
    let scl = p.PIN_15;
    let mut i2c = i2c::I2c::new_async(instance, scl, sda, Irqs, Config::default());

    info!("init tmp117");
    Timer::after(Duration::from_secs(1)).await;

    info!("Read tmp117 DEVICE_ID register");
    loop {
        let mut val = [0];

        i2c.write_read(0x48, &[0x0f], &mut val).await.unwrap();

        info!("Read value {:02x}", val[0]);
        Timer::after(Duration::from_secs(1)).await;
    }
}
```

It also appears that some of the pin traits don't match the pinout from the official documentation (https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html), for example pins 20 and 19 appears to be swapped.

Dirbaio, 2023-08-03:
perhaps you're confusing Pico and RP2040 pin numbers? For example Pico pin number 19 is RP2040 pin 14 (aka `GP14`, or `p.PIN_14`), which is I2C1 SDA.

felipebalbi, 2023-08-03:
> perhaps you're confusing Pico and RP2040 pin numbers? For example Pico pin number 19 is RP2040 pin 14 (aka `GP14`, or `p.PIN_14`), which is I2C1 SDA. The traits look good to me.

that was exactly the case. For some reason I thought `embassy-rp` referred to the pico board, but it's an rp2040 hal instead (which makes sense). After fixing that, I can communicate with the sensor.