embassy-rs/embassy #1744: embassy_nrf: Implement embedded-io async Read & Write for UarteRX and UarteTx

vilbergs, 2023-08-04:
Hello and thank you for Embassy!

I'm somewhat new to the embedded scene and trying to learn how to use Embassy for LoRa applications.

I'm using https://github.com/drogue-iot/rak811-at-driver, which requires the async Read and Write traits to be implemented for the UART port handle that gets passed to the driver. 

I'm making slow progress to implement a wrapper type around `embassy_nrf::uarte` on my own, but I was wondering if this is something that should be solved upstream instead, as the RX and TX structs already have a very similar `read` and `write` trait.

I'm happy to take a stab at this if this sounds reasonable, and if not I hope I can learn something from the discussion.

vilbergs, 2023-08-04:
I see not that `embassy_nrf::buffered_uarte` implements these traits, is there a reason `uarte` doesn't?

Dirbaio, 2023-08-10:
`uarte` is unbuffered, this means it can only receive while the task is actively waiting on a `.read().await` call. Data that's received between read calls is lost. The `embedded_io::Read` contract requires that data is not so easily lost, drivers using it assume that. So even if we'd impl Read for the unbuffered uarte, it wouldn't work well.