embassy-rs/embassy #1757: Using cyw43, join_wpa2 waits forever

massimiliano-mantione, 2023-08-06:
I am experimenting with Embassy on a Pi Pico W for a robotics project.

I have created a repo where I copied all the unpublished Embassy crates so I can build and experiment freely: "https://github.com/massimiliano-mantione/rustlab-2023-robotic-sumo".

Anyone with a Pi Pico W can test this instantly (no probe needed, I use USB logging and `cargo run` is configured to use `elf2uf2-rs`).

I am having problems connecting to a  wifi network.

The hardware works, I checked with micropython and it can always connect.

With my code, while troubleshooting, I noticed that if I remove pieces of code *after* the code that connects to the wifi, it works (it joins the network successfully).
Adding code that will actually not be executed (it is *after* the network join!), the call to `join_wpa2` will wait forever.

As an example, consider this program: "https://github.com/massimiliano-mantione/rustlab-2023-robotic-sumo/blob/2b231a9bfd8cb8ef0188144c10ed5e6c0c5d069c/bot/src/bin/wifi_tcp_client_usb_log.rs".

Find the comment in lines 238-239 ("Comment this block...").
Commenting the subsequent block, `join_wpa2` will succeed.
Leaving the code in place, `join_wpa2` will wait forever.

I tried playing with buffer sizes and network stack resources, but could not change this behavior.

Any ideas?

Dirbaio, 2023-08-06:
This is usually wrong network name, wrong password, or that the network is not actually wpa2.

Can you post trace-level logs? with `DEFMT_LOG=trace`

massimiliano-mantione, 2023-08-07:
Thanks for your kind reply, I got the issue.

In my setup I do not use defmt, and I do not use a probe: I upload the code with `elf2uf2-rs` and use the USB logger.
However, I still had defmt enabled everywhere (just because the code samples I started from defined it by default).

It turns out that expanding a defmt macro without an attached probe (like a plain `info!`) has "bad consequences".
I did the following:
- disable defmt everywhere
- for now, strictly use `log::info!` (the plain `info!` macros imported with `use defmt::*;` are now no longer in scope)

This way the connection succeeds.

The weird behavior I was observing (removing a piece of code *after* the failure point made the connection succeed) was due to the fact that the piece of code in question was invoking a defmt macro.

All in all, this has obviously been my fault: I had a setup with defmt enabled and I am not using a probe, and I was unaware of the consequences.

It would likely be nice if the rp2040 "USB logging" examples made it clear that they should be run without defmt.
Ideally they should build without defmt by default, to make the point clearer.
I know it's not simple at all: the whole `rp` examples folder shares the same build setup.
Maybe the USB logging examples should be moved to a separated, parallel `rt-on-usb` folder, with a build setup that disables defmt?
(and a README that explains the issue)

In the end, USB logging is pointless if you have a probe.

If you'd approve this setup I could make a pull request for it.
Anyway, _this_ issue is closed.
Thanks!

closed 2023-08-07 by massimiliano-mantione