embassy-rs/embassy #1759: Always make channels Copy

darkwater, 2023-08-08:
The current implementation for the embassy-sync channel structs look like:

```rust
/// Send-only access to a [`Channel`] without knowing channel size.
#[derive(Copy)]
pub struct DynamicSender<'ch, T> {
    channel: &'ch dyn DynamicChannel<T>,
}

impl<'ch, T> Clone for DynamicSender<'ch, T> {
    fn clone(&self) -> Self {
        DynamicSender { channel: self.channel }
    }
}
```

The manual Clone impl makes them Clone regardless of whether T is Clone, however Copy is still derived automatically, adding a T: Copy bound. Is there a reason for this, or can we make these all manual impls as well?

Dirbaio, 2023-08-08:
Yes, Copy should be a manual impl. Nice catch! PR's welcome.

darkwater, 2023-08-08:
See #1760 

I saw the same pattern for pipes so I did the same thing there

closed 2023-08-08 by darkwater