embassy-rs/embassy #1776: embassy-lora `GenericSx126xInterfaceVariant` equivalent for Sx127x Boards

andres-hurtado-lopez, 2023-08-13:
I'm trying to understand if the `GenericSx126xInterfaceVariant` in order to put to work for the Sx127x LoRa boards with a rp-pico, so far no luck. I'm pretty sure that the lora-phy has the struct  SX1276_7_8_9 implementation. Being  said that, I can't see a way to couple it with the embassy-lora interface. 

Is this caused by embassy-lora  missing a declaration of a similar `GenericSx127xInterfaceVariant` or just is my lack of understanding on how to properly use the existing trait with the available implementations over lora-phy ?

ceekdee, 2023-08-13:
So far the Stm32l0InterfaceVariant is the only example of an InterfaceVariant involving an Sx127x.  It is used for an stm32l0 board with an embedded sx1276.  You can model your InterfaceVariant implementation off that.  If it seems like the new InterfaceVariant may be useful across MCU/sx127x board combos, go ahead and name it GenericSx127xInterfaceVariant.  Otherwise, give it a name that allows future users to identify the MCU/sx127x board combo your InterfaceVariant represents.

If it turns out that the Stm32l0InterfaceVariant meets your needs, please still create a GenericSx127xInterfaceVariant version.  I currently want to keep the STM InterfaceVariants separate.

Examples of usage of the Stm32l0InterfaceVariant with an sx1276 board are located in the stm32l0 Embassy examples with names starting with "lora".

andres-hurtado-lopez, 2023-08-13:
I'll give it try to use `Stm32l0InterfaceVariant`.  In case it works I ĺl try to follow your advice on isolating the implementation with the combo. If in the application seems useful to the public the I'll try to do a pull request. 

Thanks for the advice.

closed 2023-08-13 by andres-hurtado-lopez