embassy-rs/embassy #1791: NoRoute error using the Pico-W on wifi

InfinityDevTech, 2023-08-17:
Hello!

Im trying to connect to my Node TCP server on port `80` with ip addr of `192.168.0.2`, but i keep getteing a NoRoute error, here is the relevant code.

Im pretty new to embedded development, so go a little easy on me lol

```rust
let fw = include_bytes!("../cyw43-firmware/43439A0.bin");
    let clm = include_bytes!("../cyw43-firmware/43439A0_clm.bin");

    let pwr = Output::new(peripherals.PIN_23, Level::Low);
    let cs = Output::new(peripherals.PIN_25, Level::High);
    //let white_led = Output::new(p.PIN_15, Level::Low);
    let mut c: embassy_rp::pwm::Config = embassy_rp::pwm::Config::default();
    c.top = 0x8000;
    c.compare_b = 8;
    let white_led_pwm = Pwm::new_output_a(peripherals.PWM_CH7, peripherals.PIN_14, c.clone());
    let mut pio = Pio::new(peripherals.PIO0, Irqs);
    let spi = PioSpi::new(
        &mut pio.common,
        pio.sm0,
        pio.irq0,
        cs,
        peripherals.PIN_24,
        peripherals.PIN_29,
        peripherals.DMA_CH0,
    );
    let state = make_static!(cyw43::State::new());
    let (net_device, mut control, runner) = cyw43::new(state, pwr, spi, fw).await;
    unwrap!(spawner.spawn(init::wifi_task(runner)));
    control.init(clm).await;
    control
        .set_power_management(cyw43::PowerManagementMode::PowerSave)
        .await;
    control.gpio_set(0, true).await;

    let config = Config::dhcpv4(Default::default());
    let seed = 0x0123_4567_89ab_cdef;

    // Init network stack
    let stack = make_static!(Stack::new(
        net_device,
        config,
        make_static!(StackResources::<4>::new()),
        seed
    ));

    loop {
        //control.join_open(WIFI_NETWORK).await;
        match control.join_wpa2(WIFI_NETWORK, WIFI_PASSWORD).await {
            Ok(_) => {
                info!("Joined the network! {}", WIFI_NETWORK);
                break;
            }
            Err(err) => {
                info!("join failed with status={}", err.status);
                flash_onboard(control).await;
                return;
            }
        }
    }

    unwrap!(spawner.spawn(init::net_task(stack)));

    let mut rx_buffer = [0; 4096];
    let mut tx_buffer = [0; 4096];
    let mut buf = [0; 4096];

    loop {
        let mut socket = TcpSocket::new(stack, &mut rx_buffer, &mut tx_buffer);
        socket.set_timeout(Some(Duration::from_secs(10)));

        info!(
            "Attempting connection to {:?} on port: {}",
            SERVER_IP, SERVER_PORT
        );
        if let Err(e) = socket
            .connect(IpEndpoint::new(SERVER_IP, SERVER_PORT))
            .await
        {
            error!("Failed to connect, this is a fatal error! {}", e);
            //error!("WIFI connection state {}", control.);
            //led::timer_fade_led(white_led_pwm, c, 5).await;
            continue;
        }
```

Dirbaio, 2023-08-17:
you need to wait until DHCP grabs an IP, which is usually ~10 seconds. Add some delay between connection attempts so the other tasks have a chance to run, then it should succeed once DHCP is done.

Or you can wait for DHCP with code, like this https://github.com/embassy-rs/embassy/blob/main/tests/rp/src/bin/cyw43-perf.rs#L91-L95

InfinityDevTech, 2023-08-17:
Ah, thanks!

I have never done something like this before so I'm just confused, I found another way to make it work by doing some weird ipv4_static hack.

Just so I know and since I would like to learn, do you have a small explanation as to why that works?

Dirbaio, 2023-08-17:
if you set a static IP, it's available instantly. DHCP takes a few seconds, during which the device has no IP address, so all attempts to connect to somewhere will fail with NoRoute.

InfinityDevTech, 2023-08-18:
Ok! Thank you!

closed 2023-08-18 by InfinityDevTech