embassy-rs/embassy #1800: IEEE 802.15.4 Support - nRF52840

alex-astronomer, 2023-08-18:
I noticed that there isn't IEEE support for the radio in the nRF52840.  I thought this issue could be used as a thread to track information and progress.

alex-astronomer, 2023-08-18:
If anyone has advice for where to get started with this, I'd be grateful to get some help!

Dirbaio, 2023-08-18:
ping @thvdveld @rubdos which I believe are working on this.

alex-astronomer, 2023-08-20:
I started some work on this, and I'm making some progress.  With those that you tagged, I'd like to connect when you can and discuss through this thread.  Thanks!

thvdveld, 2023-08-21:
@rubdos is currently on leave but will be available again later this week. I'm mainly working on the higher layers (IP and 6LoWPAN), however, I have a working nRF radio implementation lying around for testing the IP stack, but nothing fancy.

rubdos, 2023-08-21:
@alex-astronomer what parts of 802.15.4 are you interested in? I'm building a `trait Radio`, s.t. Embassy may have a vendor-independent radio stack. We'd also like to support ST and TI, for instance.

rubdos, 2023-08-21:
Also, cc @xoviat

Dirbaio, 2023-08-21:
Is it a low level "set frequency, send, receive" radio trait? 

see previous discussion about radio traits in  https://github.com/embassy-rs/embassy/issues/432 https://github.com/embassy-rs/embassy/pull/463 . It has also been tried at https://github.com/nrfconnect/sdk-nrfxlib/tree/main/nrf_802154

tldr is I'm somewhat skeptical it's possible to do non-toy implementations of complex protocols like 802.15.4 or BLE on top of hardware-agnostic radio traits. For example both Nordic's BLE SoftDevice and [802.15.4 driver](https://github.com/nrfconnect/sdk-nrfxlib/tree/main/nrf_802154) make extensive use of hardware timers, PPI, shorts to get the hardware to do things without CPU intervention with tight timing. These are very hardware-specific, if you're building the whole radio stack on top of a hardware-agnostic radio trait you can't use those hardware capabilities.

Also stm32 doesn't expose the raw radio at all, it has a blob that already implements 802.15.4 for you (I think?)

rubdos, 2023-08-22:
> Is it a low level "set frequency, send, receive" radio trait?

Not entirely. My current design looks like this:

```
pub trait Radio {
    /// Request the radio to idle to a low-power sleep mode.
    fn off(&mut self, ctx: &mut Context<'_>) -> Poll<()>;

    /// Request the radio to go in receive mode and try to receive a packet into the supplied
    /// buffer.
    ///
    /// SAFETY: The supplied buffer must remain writable until either
    /// successful reception, or the radio state changed.
    unsafe fn prepare_receive(&mut self, ctx: &mut Context<'_>, cfg: &RxConfig, bytes: &mut [u8; 128]) -> Poll<()>;

    /// Request the radio to go in receive mode and try to receive a packet.
    fn receive(&mut self, ctx: &mut Context<'_>) -> Poll<bool>;

    /// Request the radio to go in transmit mode and try to send a packet.
    ///
    /// SAFETY: The supplied buffer must remain valid until either
    /// successful reception, or the radio state changed.
    unsafe fn prepare_transmit(&mut self, ctx: &mut Context<'_>, cfg: &TxConfig, bytes: &[u8]) -> Poll<()>;

    /// Request the radio to transmit the queued packet.
    fn transmit(&mut self, ctx: &mut Context<'_>) -> Poll<()>;

    /// Returns the IEEE802.15.4 8-octet MAC address of the radio device.
    fn ieee802154_address(&self) -> [u8; 8];
}
```

I agree that it is rather challenging, but I still believe it's possible to mostly decouple the MAC-layer from the radio interface layer. I'm still rather convinced that we need *some* form of common abstraction over the different vendor interfaces. Worst case, we can still go with some dependency inversion, where the radio implementation of a chip has a generic `Mac` argument that we standardize via traits. 

I have tried a few different designs now, and I think the above might work out. The transmit and receive prepare methods accept an argument to configure the mode of transmission/reception, such that hardware-ack or most probably executor-free ack can be implemented.

> Also stm32 doesn't expose the raw radio at all, it has a blob that already implements 802.15.4 for you (I think?)

STM32 gives you options: either raw 802.15.4 MAC, Thread, BLE, and a few combinations of these. The generic interface above will only work on the raw MAC, but that's still desired for our research group (since we do research on MAC protocols on Contiki atm). Obviously, in a real-world deployment of Thread, you will use the hardware implementation with a custom `impl Driver` for STM32, but it would still be desirable to have this possibility.

Should we create an #embassy-ieee802154:matrix.org channel for futher discussion?

alex-astronomer, 2023-08-22:
> what parts of 802.15.4 are you interested in? I'm building a trait Radio, s.t. Embassy may have a vendor-independent radio stack. We'd also like to support ST and TI, for instance.

The entire motivation for this for me is to eventually be able to implement OpenThread in Rust.  I love `embassy` and I'd like to add to an existing library/framework rather than doing everything from scratch, so I ended up here.  So whatever it takes to do that is what I'm interested in :). Right now I have a toy implementation of the radio doing RX, TX, CCA, ED, and other small features, but just for the nRF52840.  That would be all I can implement as I don't have any other chips ATM, and my project is utilizing these, so I will focus on that, I think.

>  I'm somewhat skeptical it's possible to do non-toy implementations of complex protocols like 802.15.4 or BLE on top of hardware-agnostic radio traits...  ...These are very hardware-specific, if you're building the whole radio stack on top of a hardware-agnostic radio trait you can't use those hardware capabilities.

I'm trying to wrap my head around this, as I've just been focusing on a single chipset for now and haven't put much thought into others.  I think that it would be possible right?  Please be patient, as I'm not super experienced with embedded OS, and I'm intermediate with Rust...  This is my first time really experimenting with combining those two skillsets...  So from the beginning: the idea of a trait is that you are doing polymorphism by defining interfaces and common behavior.  With that understanding in mind, would it not be possible to implement the hardware-specific features in a lower level interface, that's hidden away, and then keep the common behavior that each needs to receive visible and accessible?  Like just having RX and TX and other features like @rubdos gave, and then doing everything like CCA, ED and other lower-level features behind the scenes and specific for each chipset depending on its hardware capabilities?

> I still believe it's possible to mostly decouple the MAC-layer from the radio interface layer...  ...I'm still rather convinced that we need some form of common abstraction over the different vendor interfaces.

+1 based on my limited understanding

> Should we create an #embassy-ieee802154:matrix.org channel for futher discussion?

What is the messaging app used for this project?  I'd love to get involved there and because more discussion is warranted, I'd love to get involved and do what I can to help.  I have a few nRF52840 laying around and I'm just trying to get to a point where I can get a Thread network running for a sample toy application.

---

Basically, the way that I got here is that I have a project in mind where I want to use OpenThread and the Thread network specification.  I want to utilize Rust for this, and couldn't find anything else online really for this purpose.  I started digging into the Thread spec, realized that `embassy` doesn't have a Radio HAL from what I could tell and then down the rabbit hole I went.  Basically like any good project, I started with something ultra-high level and ended up implementing a networking stack and hardware interface.  Don't you love when that happens?? :D

rubdos, 2023-08-23:
> messaging app used for this project?

Matrix: https://matrix.to/#/#embassy:matrix.org

Dirbaio, 2024-03-06:
done in https://github.com/embassy-rs/embassy/pull/2649

closed 2024-03-06 by Dirbaio