embassy-rs/embassy #1801: Properly Propagate Protocol Problems (tcp::Error -> embeded_io::Error)

avlec, 2023-08-18:
Currently there's no error translation for converting from embassy_net::tcp::Error to embedded_io::Error.
It is rather concerning to get a "Other" as an Error value instead of useful information that you have to get by breaking at the conversion spot in tcp.rs https://github.com/embassy-rs/embassy/blob/97da34595cb9943affadb1ce8785556108b977df/embassy-net/src/tcp.rs#L515

I was seeing the embedded_io Other error when under the hood the actual error was ConnectionReset, which the function above converts to Other which loses all meaning.

Dirbaio, 2023-08-18:
Agreed. This code predates adding all the variants to `embedded_io_async::ErrorKind` https://github.com/rust-embedded/embedded-hal/pull/471, that's why it's this way. PR's welcome!

(also note you can inspect the ConnectError directly, you don't need to do `.kind()`)

avlec, 2023-08-18:
Yes, I was trying to get reqwless http lib to work, stumbled across this translation error when doing a debugger to a RP Pico W. So I was using a commit from this project from 2 weeks ago, and see that the 0.4 version of embedded_io at that point only had Other. So I could only see the error from the very outside, behind an HttpClient request returning an error.

PR #1802 (2023-08-18): Add error translation to tcp errors
Translation of tpc client ConnectError and Error to the appropriate embedded_io_async errors

Closes #1801
closed 2023-08-18 by Dirbaio