embassy-rs/embassy #1819: nightly-2023-08-19 not working on aarch64-apple-darwin.

overheat, 2023-08-24:
```
❯ cargo clean
error: the 'cargo' binary, normally provided by the 'cargo' component, is not applicable to the 'nightly-2023-08-19-aarch64-apple-darwin' toolchain
```

overheat, 2023-08-24:
Change `channel = "nightly-2023-08-19"` to `channel = "nightly-2023-08-20"` in rust-toolchain.toml will fix this bug.

timokroeger, 2023-08-24:
The main branch works fine for me on a MacBook Air M1.
```
user@users-Air embassy % git pull
Already up to date.
user@users-Air embassy % cd examples/rp/
user@users-Air rp % cargo build
   Compiling bitflags v1.3.2
   Compiling critical-section v1.1.2
   Compiling stable_deref_trait v1.2.0
```

Dirbaio, 2023-08-24:
It shows as available in https://rust-lang.github.io/rustup-components-history/aarch64-apple-darwin.html

Maybe your rustup is corrupted. Please try `rm -rf ~/.rustup` to clear it out and try again.

overheat, 2023-08-28:
> It shows as available in https://rust-lang.github.io/rustup-components-history/aarch64-apple-darwin.html
> 
> Maybe your rustup is corrupted. Please try `rm -rf ~/.rustup` to clear it out and try again.

Cool, straightforward.

How many features needed that not stable? Do we have chance to use stable rustc in coming years?

Dirbaio, 2023-08-28:
it's type-alias-impl-trait and async-fn-in-trait. documented in https://github.com/embassy-rs/embassy#minimum-supported-rust-version-msrv

closed 2023-08-28 by Dirbaio