embassy-rs/embassy #1827: cyw43: Detect disassociation from infra.

Frostie314159, 2023-08-26:
It would be nice(also internally), to have a way, to detect, if we disassociated from the WiFi we were previously connected to. Be it due to signal loss or some other interference.

hirschenberger, 2023-11-09:
I also see problems when disassociated from a WIFI connection. Is there a workaround to reconnect. Where can I query if the connection still persists?

soypat, 2023-12-08:
FYI: The way we solved it in tinygo was looking for a [LINK packet with zero flag](https://github.com/soypat/cyw43439/pull/26/files#diff-450d90a83b32c33c4413143d0e63a18c79c801b963c52fb3c872b0da277aababR531-R533) to indicate a disassociation had happened and then waiting for a JOIN packet to indicate it had succesfully reconnected.