embassy-rs/embassy #1828: self-sufficient getting starting project

mark2b, 2023-08-26:
Hello. 
Is there self-sufficient "getting starting" project.
Not as part of repository with relative pathes in Cargo.toml and a lot of configurations files distributed across repository, but  self-sufficient project with all required files inside that can be compiled even without local embassy-rs repository.
I can successfully compile any example project located in repository tree, but moving it outside of repository I have a lot of compilation problems. 

Thanks
Mark

41Mo, 2023-08-27:
Hello,
you might want to look github for embassy-template/embassy-start repos.
When i explored github i found [embassy-start](https://github.com/titanclass/embassy-start) repo, which looks pretty descent.
If you would like more minimal setup you can take a look at [embassy-rp-template](https://github.com/explocion/embassy-rp-template).
On github you may find some nice fully functional projects like watermeter(if i remeber the name right).

mark2b, 2023-08-28:
Hello, @41Mo 
Thanks for great reference, especially https://github.com/explocion/embassy-rp-template
This is really what I was looking for

agausmann, 2024-04-28:
There's some community-made projects, but the couple rp2040 templates that I've found so far are out of date - they haven't been updated to point to crates.io yet. (none of the ones linked above, nor [embassy-rp-skeleton](https://github.com/SupImDos/embassy-rp-skeleton))

I think it would be helpful to have officially-maintained templates that are kept in sync with embassy releases

agausmann, 2024-04-28:
for future reference, [embassy-template](https://github.com/lulf/embassy-template) linked in [the docs](https://embassy.dev/book/dev/new_project.html) seems to be up-to-date.

closed 2026-11-06 by leftger